wasm-micro-runtime/wasm-micro-runtime #449: Other languages/ESP-IDF support question

Threadnaught, 2020-11-26:
Hi guys,

First off, thanks for putting WAMR together.

I'm working on a project called Nihilo (https://github.com/Threadnaught/Nihilo) which uses WAMR as part of a computing platform built around remote procedure calls and ECC encryption. The communication, storage and cryptography modules have all been implemented on linux so far.

I would like to run Nihilo on the ESP32, which uses the ESP-IDF and FreeRTOS. I can see that you've already done the work to support xtensa, but the IDF is not yet one of the platforms. Is this something that's on your priority list, or would I have to port it myself?

The other question is regarding other languages which compile directly into webassembly (like golang, C#, C++ etc). Is there any way to get these to use the provided c sysroot, or would a new sysroot need to be provided for each?

wenyongh, 2020-11-26:
Hi, thanks for using WAMR in Nihilo project!

For FreeRTOS, we have some experiment code for the platform, but haven't tested it yet. I just imported the code to the project, could you please help enhance it, make it work and then submit the patch back to github? Thanks a lot.

Most of the platform related APIs should have been implemented, except os_printf, os_vprintf, os_thread_get_stack_boundary and some libc APIs, see core/shared/platform/freertos folder. And there is a sample makefile source.mk under product-mini/platforms/freertos/simple for a ref. Maybe you can implement os_printf and ref source.mk to enable the product-mini simple project firstly.

For the compiler, there are many toolchains, e.g. wasi-sdk and emcc for C/C++, tinygo for go, asc for typescript, rustc/cargo for rust, please ref to the related documents for more details. Or ref to [doc/build_wasm_app.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md) for the detail usage of wasi-sdk.

Threadnaught, 2020-11-26:
FreeRTOS is not a full platform, but a collection of primitives for concurrency built on top of existing microcontroller APIs. Implementations vary, so a function that is present in one FreeRTOS-based API (like printf in ESP-IDF and Linux) may vary on another (like Serial.Println() on arduino-based platforms).

wenyongh, 2020-11-27:
Yes, seems that we should re-org the code structure, how about: (1) putting all the common APIs to folder core/shared/platform/common/freertos, these APIs can be used by all FreeRTOS-based platforms, (2) and letting the platform e.g. ESP-IDF implement the platform dependent APIs, e.g. printf? And the platform should define its own platform macro, e.g. BH_PLATFORM_ESP_IDF.

Threadnaught, 2020-11-28:
I'm investigating this fork https://github.com/nickzoic/wasm-micro-runtime (branch wamr-for-esp-idf). It has the same license so depending on its quality it could be integrated back into WAMR.

wenyongh, 2020-11-30:
Yes, there is really some useful files to ref to from the fork. But its code base is a little old, some files and codes have been updated in the latest main code base, so it might be difficult to use all its files. We re-orged the code structure in latest patch #454: (1) remove core/shared/platform/freertos folder, (2) add folder core/shared/platform/common/freertos with common-used freertos APIs, (3) and add esp-idf platform: core/shared/platform/esp-idf and product-mini/platforms/esp-idf. All the source files should have been prepared, and the CMakeLists.txt in product-mini/platforms/esp-idf was drafted ref to the fork you mentioned and the CMakeLists.txt in other platforms. We haven't compiled it yet, could you please have a look, compile it and fix the compilation issues? The efforts should be not so large as before. Thanks a lot.

Threadnaught, 2020-12-02:
I got that fork compiling for esp-idf, I am attempting to resolve some issues with it before I bringing it into this repo. I might need someone to look over the CMakeLists.txt - It's not exactly my strong suit and this one has been through three different hands.

Threadnaught, 2020-12-18:
I've been trying to get the code in the above mentioned fork working since I posted that last comment, and unfortunately I cannot; I suspect it never fully worked in the first place. I'm closing this issue for now because I need to put a lot more time into learning ESP-IDF and XTENSA before I'll be in a place to make a meaningful contribution along these lines. Thanks again for making WAMR, sorry I couldn't be more help.

closed 2020-12-18 by Threadnaught