wasm-micro-runtime/wasm-micro-runtime #464: Build wamrc AoT compiler error: "error: function cannot return function type"

JavanZhu, 2020-12-09:
HI, I'm trying to [Build wamrc AoT compiler](https://github.com/bytecodealliance/wasm-micro-runtime#build-wamrc-aot-compiler), but when running following command to build:

```bash
./build_llvm.sh
mkdir build && cd build
./cmake -DWAMR_BUILD_PLATFORM=darwin ..
make -j8
```
I got error:
```
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3126:14: error: function cannot return function type
      'int (GenericFunctionPointer, uint64 *, uint64)' (aka 'int (void (*)(), unsigned long long *, unsigned long long)')
typedef v128 (*V128FuncPtr)(GenericFunctionPointer, uint64*, uint64);
             ^
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3127:8: error: unknown type name 'V128FuncPtr'; did you mean 'VoidFuncPtr'?
static V128FuncPtr invokeNative_V128 = (V128FuncPtr)(uintptr_t)invokeNative;
       ^~~~~~~~~~~
       VoidFuncPtr
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3117:16: note: 'VoidFuncPtr' declared here
typedef void (*VoidFuncPtr)(GenericFunctionPointer, uint64*, uint64);
               ^
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3127:41: error: use of undeclared identifier 'V128FuncPtr'
static V128FuncPtr invokeNative_V128 = (V128FuncPtr)(uintptr_t)invokeNative;
                                        ^
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3127:54: error: unexpected type name 'uintptr_t': expected expression
static V128FuncPtr invokeNative_V128 = (V128FuncPtr)(uintptr_t)invokeNative;
                                                     ^
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3240:33: error: subscript of pointer to function type '__m128i_u' (aka 'int (int *)')
                    *(float32*)&fps[n_fps++] = *(float32*)argv_src++;
                                ^~~
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3248:33: error: subscript of pointer to function type '__m128i_u' (aka 'int (int *)')
                    *(float64*)&fps[n_fps++] = *(float64*)argv_src;
                                ^~~
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3258:30: error: subscript of pointer to function type '__m128i_u' (aka 'int (int *)')
                    *(v128*)&fps[n_fps++] = *(v128*)argv_src;
                             ^~~
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3261:49: error: non-object type '__m128i_u' (aka 'int (int *)') is not assignable
                    *(v128*)&stacks[n_stacks++] = *(v128*)argv_src;
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:3303:34: error: non-object type '__m128i_u' (aka 'int (int *)') is not assignable
                *(v128*)argv_ret = invokeNative_V128(func_ptr, argv1, n_stacks);
                ~~~~~~~~~~~~~~~~ ^

```

--------

build system: macOS 10.14.6
latest commit of native code:
[image]

wenyongh, 2020-12-09:
Hi, what is the target CPU, is it x86-64? We tested in MacOS x86-64, it works. It seems that the compiler doesn't recognize the v128 type in your system, could you please check whether the following codes in wasm_runtime_common.c work correctly in you system?
```C
#include <emmintrin.h>
/* unaligned */
#define v128 __m128i_u
```
If it is difficult to resolve, could you please remove WASM_ENABLE_SIMD macro or set it to 0 in wamr-compiler's CMakeLists.txt:
```bash
add_definitions(-DWASM_ENABLE_SIMD=1)
```
Thanks.
```

JavanZhu, 2020-12-09:
Hi, @wenyongh, Thank you for your reply. The target CPU of my computer is MacOS x86-64, and following code work correctly in my system.
```c
#include <emmintrin.h>
/* unaligned */
#define v128 __m128i_u
```
I found **__m128i_u** is not exist in emmintrin.h in my native computer, and after change "__m128i_u" to "__m128i", build wamrc success, I hope this information is helpful. Please let me know if you need more information, thanks.
[image]

wenyongh, 2020-12-09:
Hi, it seems that you are using some lower version of clang/llvm, could you please upgrade them to higher version, e.g. 10 or 11? Their header file defines type __m128i_u. You can ref to [https://apt.llvm.org](https://apt.llvm.org).

JavanZhu, 2020-12-10:
@wenyongh After upgraded my system to Big Sur and fixed some error, build wamrc success. Thanks again for your help.

closed 2020-12-10 by JavanZhu