wasm-micro-runtime/wasm-micro-runtime #468: Question - how to print stack trace when WASM throws exception?

AfflatusX, 2020-12-09:
I'm using AssemblyScript, when something goes wrong in the wasm logic, WAMR prints: Got exception running WASM code: Exception: unreachable.

I'm wondering if there is anyway to print full stack trace when exception is caught? Or that's something needs to be handled within AssemblyScript's runtime?

xujuntwt95329, 2020-12-10:
Hi, firstly please add `-DWAMR_BUILD_CUSTOM_NAME_SECTION=1` to enable the [call stack feature](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-debug). 

In addition, I noticed that `asc` doesn't generate `custom name section` by default, you may need to pass `--debug` option to `asc`.

AfflatusX, 2020-12-10:
ahhh @xujuntwt95329 thanks again for answer! yeah I remember seeing stack trace at one point but couldn't remember if I enabled some flag in WAMR or AS. Turned out there is something to enable on both ends. Appreciate it!

closed 2020-12-10 by AfflatusX