wasm-micro-runtime/wasm-micro-runtime #471: How to find the slowly method when execute an wasm code? (Performance)

JavanZhu, 2020-12-15:
Hi, I have developed some AssemblyScript projects and run it in wasm-micro-runtime, and found it's not very easy to inspect performance problem in my wasm code. For example

My AssemblyScript demo:
```AssemblyScript
@external("env", "native_method")
declare function native_method(
    string: ArrayBuffer
): void;

export function call_by_native(buffer_offset: i32, size: i32): void {
    var msg = String.UTF8.decodeUnsafe(buffer_offset, size);
    console.log("call_by_native: result = [" + msg + "]");
}

// main function of wasm app
export function main(): void {
    console.log("main() start");

    as_method1();

    // This method call native method, and then will receive some msg in call_by_native() method called by native method
    native_method(String.UTF8.encode(some_big_string_in_as, true));

    as_method2();

    console.log("main() finish");
}

let some_big_string_in_as = `The lyric was virtually unknown to Old English poets. Poems such as “Deor” and “Wulf and Eadwacer,” which have been called lyrics, are thematically different from those that began to circulate orally in the 12th century and to be written down in great numbers in the 13th; these Old English poems also have a stronger narrative component than the later productions. The most frequent topics in the Middle English secular lyric are springtime and romantic love; many rework such themes tediously, but some, such as “Foweles in the frith” (13th century) and “Ich am of Irlaunde” (14th century), convey strong emotions in a few lines. Two lyrics of the early 13th century, “Mirie it is while sumer ilast” and “Sumer is icumen in,” are preserved with musical settings, and probably most of the others were meant to be sung. The dominant mood of the religious lyrics is passionate: the poets sorrow for Christ on the cross and for the Virgin Mary, celebrate the “five joys” of Mary, and import language from love poetry to express religious devotion. Excellent early examples are “Nou goth sonne under wod” and “Stond wel, moder, ounder rode.” Many of the lyrics are preserved in manuscript anthologies, of which the best is British Library manuscript Harley 2253 from the early 14th century. In this collection, known as the Harley Lyrics, the love poems, such as “Alysoun” and “Blow, Northern Wind,” take after the poems of the Provençal troubadours but are less formal, less abstract, and more lively. The religious lyrics also are of high quality; but the most remarkable of the Harley Lyrics, “The Man in the Moon,” far from being about love or religion, imagines the man in the Moon as a simple peasant, sympathizes with his hard life, and offers him some useful advice on how to best the village hayward (a local officer in charge of a town’s common herd of cattle).

A poem such as “The Man in the Moon” serves as a reminder that, although the poetry of the early Middle English period was increasingly influenced by the Anglo-Norman literature produced for the courts, it is seldom “courtly.” Most English poets, whether writing about kings or peasants, looked at life from a bourgeois perspective. If their work sometimes lacks sophistication, it nevertheless has a vitality that comes from preoccupation with daily affairs.`

function as_method1() {
    // mock slow method
    for (let i = 0; i < 10000; i++) {

    }
}

function as_method2() {
    // mock slow method
    for (let i = 0; i < 10000; i++) {

    }
}

```

execute sequence:
```
main() ->
    as_method1()
    native_method() -> 
        call_by_native()
    as_method2()
```

my problem:
1. Is there any way to find the most slowly method?
2. How to calculate the total cost time of my wasm code(only include AssemblyScript, except native method)? In this demo, it equals **as_method1() + call_by_native() + as_method2()** 
3. When pass string value from native to runtime, I use following code.  But I found that encode() method has cost a lot of time when the string size is very big, so is there any way to achieve it but high performance?
```
let addr = String.UTF8.encode(some_big_string_in_as, true)
native_method(addr);
```

I know insert some log in as code is available when AssemblyScript code is very simple,. But it's impossible and complex when the AssemblyScript project is big.
Is there some solutions or suggestions to help me solve the three problems above? Thank you very much.

wenyongh, 2020-12-16:
Hi, not sure which mode do you want to apply the performance profiling, interpreter, JIT or AOT? For interpreter, it is relatively simple, we can get the execution time of a function by tracking the interpreter's stack frame: when allocating a stack frame, mark its start timestamp, and when freeing a stack frame, get its current running time (=current timestamp - start timestamp), which is the running time of the function the frame belongs to. I uploaded a patch for your reference, note that it is just a draft version, you might need to improve it by yourself.

For JIT, the JITed function symbols can be resolved after LLVM JIT compilation is applied, so we should be able to use some performance profiling tool to profile it, we recommend using the vtune tool, you might ref to [installing-intel-oneapi-toolkits-via-apt](https://software.intel.com/content/www/us/en/develop/articles/installing-intel-oneapi-toolkits-via-apt.html) to install it. Note that the JITed functions are named as "aot_func#0", aot_func#1", ...

For AOT, it is not easy to implement it as the AOTed function symbols cannot be resolved by OS.  We might get the profiler data like what interpreter does, but it needs to implement some LLVM IRs, which really takes effort. So could you please use JIT mode instead? Thanks.

[perf_profiling_interp.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5701349/perf_profiling_interp.zip)

JavanZhu, 2020-12-17:
Hi @wenyongh，the mode of my runtime is interpreter, and your reply is very helpful for me. Thank you very much.

closed 2020-12-17 by JavanZhu