wasm-micro-runtime/wasm-micro-runtime #476: failed to run aot file on mac

JavanZhu, 2020-12-23:
Hi, I follow the tutorial [build-wamrc-aot-compiler](https://github.com/bytecodealliance/wasm-micro-runtime#build-wamrc-aot-compiler) to build a wamrc, and compile the hello_world.wasm to hello_world.aot
```
 ./wamrc --target=x86_64 -o hello_world.aot hello_world.wasm
```
above command outputs:
```
Create AoT compiler with:
  target:        x86_64
  target cpu:    
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file hello_world.aot was generated.

```

Finally I am trying to execute aot file on iwasm, but got following error:
```
./iwasm ../../../../wamr-compiler/build/hello_world.aot 

ASSERTION FAILED: (uintptr_t)aot_text < INT32_MAX, at file wasm-micro-runtime/core/iwasm/aot/aot_loader.c, line 1896
[1]    92519 abort      ./iwasm ../../../../wamr-compiler/build/hello_world.aot
```

## informations
* arch: mac RELEASE_X86_64 x86_64
* wasm and aot files: 
[hello-world-wasm-and-aot-files.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5732882/hello-world-wasm-and-aot-files.zip)
* commits: [35acc7f](https://github.com/bytecodealliance/wasm-micro-runtime/commit/35acc7f3365f1701299ec0be8563c2afaa61683d)

xujuntwt95329, 2020-12-23:
Hi, your case works well on my Linux platform, but I don't have Mac environment.
Could you please modify `aot_loader.c` line 989 and 1890
https://github.com/bytecodealliance/wasm-micro-runtime/blob/cb0e593c15903827987f8a9e618f40b3e204eac1/core/iwasm/aot/aot_loader.c#L989
https://github.com/bytecodealliance/wasm-micro-runtime/blob/cb0e593c15903827987f8a9e618f40b3e204eac1/core/iwasm/aot/aot_loader.c#L1890
to 
``` C
#if !defined(BH_PLATFORM_LINUX_SGX) && !defined(BH_PLATFORM_WINDOWS) 
    && !defined(BH_PLATFORM_DARWIN)
```
besides, you may need to add `--size-level=1` to wamrc when you compile AoT module

If this works, could you please commit the change back to WAMR? Thanks a lot.

JavanZhu, 2020-12-26:
Hi, @xujuntwt95329. yes, it works. Thank you for your reply, I'll test some other aot case and commit the change back to WAMR.

closed 2020-12-26 by JavanZhu