wasm-micro-runtime/wasm-micro-runtime #477: wasm_sgx didn't run *.wasm which generated by wasm-pack

bradyjoestar, 2020-12-23:
I try to run a `wasm` file in enclave of sgx, It failed.
The wasm file is generated by `wasm-pack`, and it is writed by rust.
Here is the source code:
```
use wasm_bindgen::prelude::*;

// When the `wee_alloc` feature is enabled, use `wee_alloc` as the global
// allocator.

#[wasm_bindgen]
pub fn greet() {
    println!("Hello, wasmsgx!");
}

```
the `cargo.toml` is 
```
[package]
name = "wasmsgx"
version = "0.1.0"
authors = ["webb.shi <bradyjoestar@gmail.com>"]
edition = "2018"

[lib]
crate-type = ["cdylib", "rlib"]

[features]
default = ["console_error_panic_hook"]

[dependencies]
wasm-bindgen = "0.2.63"

# The `console_error_panic_hook` crate provides better debugging of panics by
# logging them with `console.error`. This is great for development, but requires
# all the `std::fmt` and `std::panicking` infrastructure, so isn't great for
# code size when deploying.
console_error_panic_hook = { version = "0.1.6", optional = true }

# `wee_alloc` is a tiny allocator for wasm that is only ~1K in code size
# compared to the default allocator's ~10K. It is slower than the default
# allocator, however.
#
# Unfortunately, `wee_alloc` requires nightly Rust when targeting wasm for now.
wee_alloc = { version = "0.4.5", optional = true }

[dev-dependencies]
wasm-bindgen-test = "0.3.13"

[profile.release]
# Tell `rustc` to optimize for small code size.
opt-level = "s"

```

and the error log is 
```
WASM module instantiate failed: allocate memory failed
```

Does `wamr` support rust-wasm now? If yes ,what should I do to run this demo? Thanks a lot!

wenyongh, 2020-12-23:
Hi, it seems that the heap size configured for sgx enclave is not enough, could you please cd enclave-sample, modify file Enclave/Enclave.config.xml, enlarge HeapMaxSize:
```bash
<HeapMaxSize>...</HeapMaxSize>
```
and then
```bash
make clean && make
```
and run iwasm again? Thanks.

bradyjoestar, 2020-12-24:
> Hi, it seems that the heap size configured for sgx enclave is not enough, could you please cd enclave-sample, modify file Enclave/Enclave.config.xml, enlarge HeapMaxSize:
> 
> ```shell
> <HeapMaxSize>...</HeapMaxSize>
> ```
> 
> and then
> 
> ```shell
> make clean && make
> ```
> 
> and run iwasm again? Thanks.

It worked! Really appreciate for your help! :smile:

closed 2020-12-24 by bradyjoestar