wasm-micro-runtime/wasm-micro-runtime #478: iwasm doesn't support *.wasm which is writeen by rust-lang

bradyjoestar, 2020-12-24:
platform: the `iwasm` on `linux-18.04`.

I generated the `wasmsgx_bg.wasm` by `wasm-pack`.

- I executed the `wasmsgx_bg.wasm` and it didn't print the expected log.
- However the `test. wasm` file generated by clang run correctly.

The rust code:
```
use wasm_bindgen::prelude::*;

// When the `wee_alloc` feature is enabled, use `wee_alloc` as the global
// allocator.

#[wasm_bindgen]
pub fn greet() {
    println!("Hello, wasmsgx!\n\n\n");
}

#[wasm_bindgen]
pub fn main(){
    print!("test main function");
}
```

The commands is following:
```
root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# ./iwasm test.wasm 
Hello world!
buf ptr: 0x11510
buf: 1234
a is: 14488
root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# ls
CMakeCache.txt  CMakeFiles  cmake_install.cmake  iwasm  libiwasm.so  libvmlib.a  Makefile  test.wasm  wasmsgx_bg.wasm
root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# ./iwasm wasmsgx_bg.wasm 
root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# ls
```

How could I print the log of `wasmsgx_bg.wasm`? Thank you!

bradyjoestar, 2020-12-24:
and I changed the rust code to :
```
use wasm_bindgen::prelude::*;

// When the `wee_alloc` feature is enabled, use `wee_alloc` as the global
// allocator.

#[wasm_bindgen]
pub fn greet() {
    println!("Hello, wasmsgx!\n\n\n");
}

```

Built and restarted it, here is the result:
```
root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# ./iwasm -f greet wasmsgx_bg.wasm 

root@mcdullsin-Latitude-5480:~/wamr/product-mini/platforms/linux/build# 
```

wenyongh, 2020-12-24:
@bradyjoestar I tried wasm-pack, the wasm file generated doesn't import any function, not sure how the rust println is compiled into wasm bytecode, normally the wasm file should import native function to output the log, e.g. printf/puts/writev. Could you please try cargo build instead?
```Bash
cargo build --release --target wasm32-wasi
```
The output file is under target/wasm32-wasi/release. Thanks.

bradyjoestar, 2020-12-24:
> @bradyjoestar I tried wasm-pack, the wasm file generated doesn't import any function, not sure how the rust println is compiled into wasm bytecode, normally the wasm file should import native function to output the log, e.g. printf/puts/writev. Could you please try cargo build instead?
> 
> ```shell
> cargo build --release --target wasm32-wasi
> ```
> 
> The output file is under target/wasm32-wasi/release. Thanks.

It worked. Thanks for your help!  :smile:

closed 2020-12-24 by bradyjoestar