wasm-micro-runtime/wasm-micro-runtime #482: Allow stdout/stderr output redirection

MaxDesiatov, 2020-12-28:
We currently use WAMR as a library embedded in a developer tool. We need to run abirtrary WASI binaries and redirect their output to the dev tool for parsing. Unfortunately, at the moment WAMR redirects all WASI stdout/stderr output to the corresponding host stdout/stderr. It would be amazing if there was a way to redirect those outputs to some arbitrary file, pipe, or store that in-memory at a provided address.

wenyongh, 2020-12-29:
Hi, the stdout and stderr are file descriptors inserted to wasi fd table in advance when init wasi context by function wasm_runtime_init_wasi:
```C
fd_table_insert_existing(curfds, 1, 1)
fd_table_insert_existing(curfds, 2, 2)
```
And normally the wasi toolchain compiles the log output from wasm app source to wasm app bytecode by calling wasi api *fd_wirte* or *fd_pwrite* with stdout/stderr file descriptor, for example, wasi-sdk/emcc compiles C printf/puts into wasi api fd_write, and eventually the function writev, pwritev or pwrite are called in function wasmtime_ssp_fd_write or wasmtime_ssp_fd_pwrite in file core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:
```C
__wasi_errno_t wasmtime_ssp_fd_write(...)
{
  ...
  ssize_t len = writev(fd_number(fo), (const struct iovec *)iov, (int)iovcnt);
  ...
}
```
So if you want to output the log by yourself, you might re-implement the code, e.g.:
```C
__wasi_errno_t wasmtime_ssp_fd_write(...)
{
  ...
  ssize_t len = writev(fd_number(fo), (const struct iovec *)iov, (int)iovcnt);
  if (fd_number(fo) == 1 || fd_number(fo) == 2) {
    int i;
    const struct iovec *iov1 = (const struct iovec *)iov;
    for (i = 0; i < (int)iovcnt; i++, iov1++) {
      if (iov1->iov_len > 0 && iov1->iov_base) {
        char buf[16];
        snprintf(buf, sizeof(buf), "%%.%ds", (int)iov1->iov_len);//make up format string "%.ns"
        printf(buf, iov1->iov_base);
      }
    }
  }
...
}
```
Not sure what your wasi toolchain is, could you please test whether it works? Thanks.

wenyongh, 2021-03-21:
The wasi stdout output redirect feature is implemented, please ref to [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#wamr_bh_vprintfvprintf_callback-default-to-disable-if-not-set) for more details.

closed 2021-03-21 by wenyongh