wasm-micro-runtime/wasm-micro-runtime #483: performance test for opencv4.2.0 cvtColor

wolfplain, 2020-12-29:
when I did a performance test for opencv based on WASMR AOT and JIT mode, the results ware not meeting the expectation, which is **10 times slower** than native;
test results:
**Android native（based on one project） | 0.701ms
Android WAMR JIT | 136.52ms
Android WAMR AOT（WASI-SDK） | 6.58ms**
code cal_times is 100:
```
int opencv_cvtColor(int cal_times, uint8_t* img_in, int width, int height) {
    unsigned char* pixels = img_in;
    cv::Mat im((int)height, (int)width, CV_8UC4, pixels);
    cv::Mat detectMat;
    clock_t start_t = clock();
    for (int i = 0; i < cal_times; i++) {
        cv::cvtColor(im, detectMat, cv::COLOR_RGBA2RGB);
    }
    int end_t = clock() - start_t;
    return end_t;
}

```

wenyongh, 2020-12-29:
Hi, it is interesting, could you please give more details about how to build the opencv workload to wasm and how to reproduce the issue? For the AOT performance, I guess that the native compilation enables the simd128, avx512, or pthread feature while the wasm compilation doesn't. Note that simd128 and avx512 should be enabled by gcc by default. For wasm toolchain, e.g. wasi-sdk or emsdk, please add -msimd128 option to enable simd, and for wamrc, please add --enable-simd option, for iwasm, please add -DWAMR_BUILD_SIMD=1.

We updated some of our latest workload samples to use the latest wasi-sdk-12.0 toolchain, with the simd headers copied from emsdk and with -msimd128 option, we can build these workloads by wasi-sdk-12.0 with simd enabled. Could you please have a try?

For the JIT performance issue, I guess that the compilation time to wasm to native code is a little long, so the performance looks poor.

wolfplain, 2020-12-30:
Thanks for your suggestions.
when I build wasm add -DWAMR_BUILD_SIMD=1 on linux version, it can get better performance, but as I known for SIMD on Android, currently WAMR not support yet. how about the support plan, can you share me some details.

about build the opencv workload， you can build bitcode libraries according to the office guild line [https://docs.opencv.org/master/d4/da1/tutorial_js_setup.html](https://docs.opencv.org/master/d4/da1/tutorial_js_setup.html)
like `python ./opencv/platforms/js/build_js.py build_js --build_wasm --simd`, 
it will generate some bitcode libraries, and they can be used by wasm;

wenyongh, 2021-07-30:
@wolfplain do you mean SIMD for aarch64 in Android? Currently SIMD for both x86-64 and aarch64 has been implemented, we tested it on Linux but haven't tested it on Android yet, normally it should be supported on Android, could you please modify the android's CMakeLists.txt by yourself (set (WAMR_BUILD_TARGET "X86_64"), set (WAMR_BUILD_AOT 1)), and use cmake .. -DWAMR_BUILD_SIMD=1 to test it? Thanks.

WANG-Bo-Wen, 2021-11-09:
> Thanks for your suggestions. when I build wasm add -DWAMR_BUILD_SIMD=1 on linux version, it can get better performance, but as I known for SIMD on Android, currently WAMR not support yet. how about the support plan, can you share me some details.
> 
> about build the opencv workload， you can build bitcode libraries according to the office guild line https://docs.opencv.org/master/d4/da1/tutorial_js_setup.html like `python ./opencv/platforms/js/build_js.py build_js --build_wasm --simd`, it will generate some bitcode libraries, and they can be used by wasm;
> 
> @wolfplain  Could you please provide some detail about how to use opencv's wasm lib  in WAMR? I got the bitcode lib of opencv like opencv_js.wasm and opencv.js,  but it doesn't work when I use this wasm lib in WAMR. Thank you.

woodsmc, 2026-08-15:
Looks like this issue is not reproducible, the original author has not come back with feedback following replies. I'm going to mark this as closed.

closed 2026-08-15 by woodsmc