wasm-micro-runtime/wasm-micro-runtime #489: How to print crash stack trace of AssemblyScript in Aot Mode

JavanZhu, 2020-12-31:
In intercept mode, we can use function https://github.com/bytecodealliance/wasm-micro-runtime/blob/f2a63d8623c3517a5ce6b10b46f92182aa351fa5/core/iwasm/interpreter/wasm_runtime.c#L2200 to print crash stack of webassembly code like this:
```
#00 ~lib/rt/pure/decrement 
#01 ~lib/rt/pure/__release 
#02 debug_unreachable/main
```
But it's not works in Aot Mode, how can we print crash stack trace of AssemblyScript when execute aot file? thank you.

xujuntwt95329, 2020-12-31:
Hi, currently stack trace for AoT is not supported, the call stack of AoTed function is not managed by the runtime, so it's not easy to get this information.

Maybe you can run your code in JIT mode and use gdb/lldb to track, but it is really a hard job.

JavanZhu, 2020-12-31:
Hi @xujuntwt95329, our wasm code is running on wasm-micro-runtime with aot mode now. Currently stack trace is important for us, because developer can use it to locate the error of their code.  So Is there any support plan for AOT stack trace, or could you please give me any tips about how to achieve it, thank you very much.

xujuntwt95329, 2021-01-04:
Hi, we had a discussion on this topic, there are two main difficulties in implementing AoT call stack:
1. The stack frame of AoTed functions are managed directly by the system, we can't get these information
2. To reduce the footprint, we didn't reserve function symbols in AoT module, which makes it impossible to get the function name during run time

For your requirement, at least the following things need to be done:
- During generating AoT file, add custom name section to the AoT module
- Record these names from name section during loading time (store in AoTModule)
- Add a runtime API `aot_alloc_frame(WASMExecEnv *exec_env, const char *func_name)`, this API should create a structure (we can call it an `auxiliary stack frame`) which stores the current function name
- During AoT compilation, when you encountered a `call` instruction in wasm code section, we need to add a call to `aot_alloc_frame` API firstly to generate the auxiliary stack frame. Of course, you also need to release the current auxiliary stack frame when function returns.
- When an exception occurs, you can then get the call stack from the auxiliary call stack.

> Note: this is not an approach design, it's just some possible tips for discussion. We haven't planned to implement this yet.

JavanZhu, 2021-01-07:
Thank you very much for your reply, I will try it according to your suggestion.

closed 2021-01-07 by JavanZhu

xujuntwt95329, 2021-01-18:
Hi, @JavanZhu 

We've provided the AoT call stack feature in recent PR #501, you can have a try.

However, this is a simple approach, we didn't parse the `custom name section` in AoT mode. Instead, we extract the function name from the `export section`. For `wasi-sdk`, you can add `-Wl,--export-all` to export all functions in the wasm module, and add `--enable-dump-call-stack` to `wamrc` to generate AoT module with auxiliary stack frame support. I don‘t know if there is a similar option for `asc`. If the function is not exported, we will print the function index.

Please refer to [Enable dump call stack feature](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#enable-dump-call-stack-feature) for more details.

JavanZhu, 2021-01-20:
> Hi, @JavanZhu
>
> We've provided the AoT call stack feature in recent PR #501, you can have a try.
>
> However, this is a simple approach, we didn't parse the `custom name section` in AoT mode. Instead, [...]

HI @xujuntwt95329 Thank you very much and I'll have a try.