wasm-micro-runtime/wasm-micro-runtime #491: Is App Manager / App Framework supported on Android?

AfflatusX, 2021-01-04:
I'm trying to build WAMR with App Framework on Android, but it complains about app_manager_timer_stop and app_manager_timer_start not defined.

Looks like those functions are only defined in Linux and Zephyr platform. Is app framework not supported on Android?

xujuntwt95329, 2021-01-04:
Hi, you can add a folder `Android` under `core/app-mgr/app-manager/platform`, and implement the required functions. You can remain empty implementation just like the Linux platform to pass the compilation.

AfflatusX, 2021-01-04:
I see. Thanks for the reply! If I leave them unimplemented, would app manager/timer API still work? Looks like those functions are used by the watchdog timer but I'm not sure what exactly is that for.

xujuntwt95329, 2021-01-05:
App manager / timer APIs will still work, we implement the timer APIs under `core/shared/utils/runtime_timer.c`. 
Watchdog timer is used to monitor the status of the app, it will not affect the execution logic.

AfflatusX, 2021-01-05:
Gotcha @xujuntwt95329 ! Adding dummy stub worked. Closing the issue.

closed 2021-01-05 by AfflatusX