wasm-micro-runtime/wasm-micro-runtime #496: Aot high probability load failed in Android Platform

newbeeXin, 2021-01-13:
I convert a relatively complex WASM application to the AOT file by using the Warmc, the AOT file size is 4 MB.  When i use the wasm_runtime_load function to load the AOT file in Android Platform (aarch64 architecture), there is a high probability that the error message "wasm_runtime_load aot failed AOT module load failed: target address out of range." is displayed. But it runs normally sometimes when loading the same AOT. I can't find the reason.
And this problem does not occur when a simple WASM application (like Fibonacci sequence test)  is converted to an AOT file(15KB).

wenyongh, 2021-01-14:
Hi, the wasm loader fails to apply relocation as the target address doesn't meet the requirement, e.g. target_addr isn't in the range of [cur_addr - 128M, cur_addr + 128M], not sure whether it is caused by the LLVM size opt level, could you please try adding --size-level=1 when running wamrc? Thanks.

newbeeXin, 2021-01-14:
Thank you very much. When I add size-level=1 , I run WAMRC to compile Aarch64 architecture AOT, the error message "LLVM ERROR: Only small, tiny and large code models are allowed on AArch64" is displayed. Then, I set size-level=0, the compilation is normal and the aot is loaded properly on the runtime, the error message "target address out of range" is not displayed. I think it is caused by the LLVM code size opt level!

closed 2021-01-18 by newbeeXin