wasm-micro-runtime/wasm-micro-runtime #499: About LIBC-WASI supported on windows

bxkai, 2021-01-17:
I try to use LIBC_WASI to build windows failed, I think it's not supported on window. I would like to ask a few questions:
1. Is there a plan to support wasi on windows?
2. Whether the  LIBC_WASI's implemention sandboxed-system-primitives can support windows? I tried to fix it in windows and found that it is very difficult
3. I want to use [uvwasi](https://github.com/nodejs/uvwasi)  to support WASI on windows. Is it feasible?

Thank you!

wenyongh, 2021-01-19:
Hi, currently we have no plan to support wasi on Windows yet, and yes, it might be difficult to port sandboxed-system-primitives library to Windows. We have a glance at uvwasi, it seems that it implements all the WASI APIs, and may be a good option to support WAMR Windows WASI if it can be ported to Windows. I upload a patch to add the build framework for uvwasi, could you please try to port the uvwasi based on it by yourself?
Note that the patch supposes that the uvwasi source code is cloned under core/deps, and you can build it in Linux platform:
```bash
cd product-mini/platforms/linux/
mkdir build && cd build
cmake .. -DWAMR_BUILD_LIBC_UVWASI=1
make
```
There are some compilation errors on uvwasi, you need to resolve it by yourself. To port it to Windows, there are some steps you need to do:
- Define macro WASM_ENABLE_LIBC_WASI=1 and WASM_ENABLE_UVWASI=1
- Initialize uvwasi on wasm_runtime_common.c, function wasm_runtime_init_wasi
- Implement WASI APIs in core/iwasm/libraries/libc-uvwasi/libc_wasi_wrapper.c, which should call uvwasi APIs and before call them, you might need to convert some of the arguments from wasm app space to native space to fit uvwasi API's requirements, you can ref to current libc-wasi/libc_was_wrapper.c for more details

And could you please also commit the patch back to WAMR repo if it is finished? Thanks.
[wamr_uvwasi.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5833192/wamr_uvwasi.zip)

bxkai, 2021-01-31:
@wenyongh Hi，I implement uvwasi on your patch, which can work normally in windows. The code has been  been uploaded to my fork branch。[The changes ](https://github.com/bxkai/wasm-micro-runtime/commit/fc8ba7f3039b5c58c3f77920b517473f0751531f)are as follows：
1. Add uv lib on libc_ uvwasi.cmake  
2. Modify invokeNative_em64.asm，solves the problem that when wasm calls native api on win64, the fourth parameter is 0.

wenyongh, 2021-02-01:
@bxkai cool, thanks a lot! Could we have more tests on the patch and merge it into the main repo? Thanks.

bxkai, 2021-02-04:
@wenyongh  Yes，of corse！

xujuntwt95329, 2021-02-23:
Hi @bxkai,

We've submitted a PR #534 to (experimentally) support uvwasi, and it is enabled by default on Windows platform.

Paril, 2021-02-23:
I tested the new UV pushes with a new CMake build, but the `_deps\libuv-src\include` directory isn't put in the Include Directories for MSVC, so it doesn't seem to build by default. I'll need to poke around to see why it's not in there.

xujuntwt95329, 2021-02-23:
Have you re-run `cmake ..` in your build path? the libuv source should be automatically downloaded by cmake process.

Paril, 2021-02-23:
It is, but there is no "uvwasi.h". There is "uv.h"

xujuntwt95329, 2021-02-23:
Seems you are missing uvwasi source, you need to `git clone` the uvwasi repo into `core/deps` firstly.

Please refer to [build_wamr.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-libc)

Paril, 2021-02-23:
Oh I see, libuv is separate from uvwasm. I missed this step, thanks!

xujuntwt95329, 2021-02-23:
The include path is added by [libc_uvwasi.cmake](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-uvwasi/libc_uvwasi.cmake)

Maybe you need to re-run `cmake ..` and confirm the `DWAMR_BUILD_LIBC_UVWASI` flag is set

Paril, 2021-02-23:
Yeah, I've wiped everything and it's still not adding the include path into vmlib for some reason. I can confirm that BUILD_LIBC_UVWASI is set, and that `UVWASI_DIR` is pointing to the correct folder.

Paril, 2021-02-23:
Wait, my bad. I was right, it's not in `vmlib`'s AdditionalIncludeDirectories.

```
      <AdditionalIncludeDirectories>G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\interpreter;G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\aot;G:\Quake 2\wasm\wasm-micro-runtime\product-mini\platforms\windows\build\_deps\libuv-src\include;G:\Quake 2\wasm\wasm-micro-runtime\core\deps\uvwasi\include;G:\Quake 2\wasm\wasm-micro-runtime\product-mini\platforms\windows\..\..\..\core\iwasm\include;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\platform\windows;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\platform\windows\..\include;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\mem-alloc;G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\common;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\utils;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\utils\uncommon;%(AdditionalIncludeDirectories)</AdditionalIncludeDirectories>
```

For comparison, here's iwasm's project, which does include it:

```
      <AdditionalIncludeDirectories>G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\interpreter;G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\aot;G:\Quake 2\wasm\wasm-micro-runtime\product-mini\platforms\windows\build\_deps\libuv-src\include;G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\libraries\libc-uvwasi\..\..\..\deps\uvwasi\include;G:\Quake 2\wasm\wasm-micro-runtime\product-mini\platforms\windows\..\..\..\core\iwasm\include;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\platform\windows;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\platform\windows\..\include;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\mem-alloc;G:\Quake 2\wasm\wasm-micro-runtime\core\iwasm\common;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\utils;G:\Quake 2\wasm\wasm-micro-runtime\core\shared\utils\uncommon;%(AdditionalIncludeDirectories)</AdditionalIncludeDirectories>
```

Paril, 2021-02-23:
Okay, so here's the first major issue I'm running into.

Executing `wasm_runtime_set_wasi_args` causes a WASI initialization failure. I've nailed down the issue to uvwasi seemingly just erroring if stdout/in/err are unavailable, which is the case I'm in because I'm not running WAMR directly from an executable, I'm running it through a DLL loaded from an EXE. I'll have to poke uvwasi about this one.

bxkai, 2021-02-27:
@Paril @xujuntwt95329 Windows applications without console do not have stdin, stdout, stderr, which cause to uvwasi init failed. Is it possible to add  interface for users (for example, wasm_runtime_set_wasi_args) to set the three file descriptors stdin / stdout / stderr？And let users replace in / out / err with open file descriptors.

/* Setup the initialization options. */
  init_options.in = 0;
  init_options.out = 1;
  init_options.err = 2;

Paril, 2021-02-27:
I filed an issue with uvwasi about this. It just hard-errors if they aren't available.

wenyongh, 2021-03-21:
Libc WASI support for Windows was implemented based uvwasi, please ref to [build iwasm on Windows](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#windows), and the wasi stdout redirection was implemented, please ref to [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#wamr_bh_vprintfvprintf_callback-default-to-disable-if-not-set) for more details.

closed 2021-03-21 by wenyongh

Paril, 2021-03-21:
I can confirm uvwasi works well enough for Windows. Thank you for the work!