wasm-micro-runtime/wasm-micro-runtime #502: Run WASM-MICRO-RUNTIME on cortex-m7

Rashow, 2021-01-18:
Hello guys and girls,
thank you for your good work on this runtime.
However, i cant quite understand the porting guide to get this micro-runtime running on a Cortex-m7.
The Porting guide dont tell me which files/folders exactly i need to include to my project. 
I think i included the wasm-micro-runtime wrong on my system. i think i didn't do it quite well, so i decide to start from beginning asking you guys now how to do it right.
I use segger-embedded-studio to build my projects and i want the runtime to load a aot-wasm module generated by wasmrc. 
It would be nice if someone helps me which files i exactly need to include to my project and which flags i need to run. I use a cortex-m7. There is a linux like OS installed. I dont have a MMU so i want to use a buffer as memory for malloc.

xujuntwt95329, 2021-01-19:
Hi,

> The Porting guide dont tell me which files/folders exactly i need to include to my project.
I think i included the wasm-micro-runtime wrong on my system. i think i didn't do it quite well, so i decide to start from beginning asking you guys now how to do it right.

Seems you are not using our cmake build system, if you need to collect the source code manually, you can refer to the Makefile  from some platform. Here is the Makefile from `product-mini/platforms/alios-things/aos.mk`:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/6847e15f64a6d8c333ba92225425f7c6932a3d30/product-mini/platforms/alios-things/aos.mk#L63-L114

The `GLOBAL_INCLUDES` are the include dirs, the `$(NAME)_SOURCES` is the source code you need, you may need to change some platform/arch related files according to your platforms.

> I dont have a MMU so i want to use a buffer as memory for malloc.

To use a global buffer as memory, please use WAMR's `Alloc_With_Pool` mode:
``` C
static char global_heap_buf[512 * 1024];
RuntimeInitArgs init_args;
memset(&init_args, 0, sizeof(RuntimeInitArgs));

/* configure the memory allocator for the runtime */
init_args.mem_alloc_type = Alloc_With_Pool;
init_args.mem_alloc_option.pool.heap_buf = global_heap_buf;
init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);
```

Please refer to [runtime initialization](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md#the-runtime-initialization) for more details.

As for the build flags, please refer to [cmake building configurations](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#wamr-vmcore-cmake-building-configurations)
The `WAMR_BUILD_XXX` are CMake flags, the corresponding C flags are `WASM_ENABLE_XXX` (except BUILD_TARGET_XXX)

These flags may be useful for your platform:
- BUILD_TARGET_ARM
- WASM_ENABLE_INTERP=1
- WASM_ENABLE_AOT=1
- WASM_ENABLE_LIBC_BUILTIN=1
- WASM_ENABLE_LIBC_WASI=1

Rashow, 2021-01-21:
Tank you very much. It almost worked.
Now i get an error called ""AOT module load failed: invalid target type, expected thumb.*_vfp but got thumbv4t"
Im sure that either i defined by build_target wrong or i build my aot file wrong.
My defines:
`BH_MALLOC=wasm_runtime_malloc
BH_FREE=wasm_runtime_free
BUILD_TARGET_THUMB_VFP
WASM_ENABLE_INTERP=1
WASM_ENABLE_AOT=1
WASM_ENABLE_FAST_INTERP=0
WASM_ENABLE_LIBC_BUILTIN=1
WASM_ENABLE_LIBC_WASI=0
LIBC_BUILTIN_DIR="./wasm-micro-runtime/core/iwasm/libraries/libc-builtin"
WASM_ENABLE_LIBC_BUILTIN=1
WAMR_BUILD_TARGET="THUMB.*_VFP"
BUILD_TARGET=WAMR_BUILD_TARGET`

the way i build my aot file:
`wamrc --target=thumbv4t -o wasmapp.aot /root/wasmapp.wasm`

I hope you can help me again. Thank you in advance.

wenyongh, 2021-01-23:
Hi, the format of WAMR_BUILD_TARGET for thumb should be [\<arch\>\[\<sub-arch\>\]\[_VFP\]](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-platform-and-architecture), e.g. THUMBV7, THUMBV7T, or THUMBV7T_VFP, normally it should not be  "THUMB.*_VFP" which is used by by cmake file to detech the string pattern.
For cortex-m7, could you please try `-DWAMR_BUILD_TARGET=THUMBV7_VFP` (or `-DWAMR_BUILD_TARGET=THUMBV7` if it doesn't work) and use wamrc with the following command:
  `wamrc --target=thumbv7 --target-abi=eabi --cpu=cortex-m7 -o wasmapp.aot wasmapp.wasm`
Ref to [little vgl sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/littlevgl) and its [zephyr-build CMakeLists.txt](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/littlevgl/vgl-wasm-runtime/zephyr-build/CMakeLists.txt) for more details. Thanks.

closed 2021-03-06 by wenyongh