wasm-micro-runtime/wasm-micro-runtime #508: Save AOT file on compilation of WASM binary using wasm-c-api

robinvanemden, 2021-01-30:
We are using the wasm-c-api successfully in several of our applications. For some of our use cases, it would be useful to be able to load a WASM, run it, save the compiled module, and then use that compiled module on every next run. 

To make that work, can I serialize a compiled module to disk in a wasm-c-api application? Ideally, as an AOT version of the original WASM binary?

wenyongh, 2021-02-01:
Hi, not sure what you want to save to disk, the WASM or AOT file which is readonly and not to be changed again, or the module instance data in memory and to be used in next time? Normally the former is called XIP (execution-in-place), and the latter is called snapshot. Both of them are really difficult to implement. For XIP, there is an [RPC issue](https://github.com/bytecodealliance/wasm-micro-runtime/issues/433) opened, we might implement the AOT XIP in the future.

For snapshot, there is no any plan yet, you might have a try by yourself. For example, in the first time run, serialize and save the module instance runtime data, including linear memory data (memory size, memory data), global data, table data, app heap data (not sure whether you can disable app heap or not). And in the next time, load AOT file again, and then restore the saved module instance data. Please refer to AOTModuleInstance/AOTMemoryInstance/ structure for more details.

robinvanemden, 2021-02-04:
Hi Wenyongh, thanks for your clarifying answer, your pointers on how we might go about implementing snapshots with the WAMR and the link to the RPC issue. In my question, I was actually refering XIP - though snapshots would be valuable to as too.