wasm-micro-runtime/wasm-micro-runtime #510: question about env.abort()

JavanZhu, 2021-02-02:
Hi, I wrote an example that using [AssemblyScript Error method](https://www.assemblyscript.org/stdlib/error.html#constructor):
```AssemblyScript
import * as console from "../../wamr_app_lib/console"

export function _start(): void {
    let map = new Map<string, string>();

    if (map.has("1")) {
        console.log("key 1 exist");
    } else {
        console.log("key 1 not exist");
        throw new Error("key 1 not exist");
    }
}
```
These producing error while running:
```
./iwasm -f _start abort_test.wasm
```
```
[20:42:47:855 - 1D459E00]: failed to check signature '(i)' and resolve pointer params for import function (env abort)

[20:42:47:913 - 1D459E00]: warning: failed to link import function (env, abort)
key 1 not exist
Exception: failed to call unlinked import function (env, abort)
```

I notice that 'new Error("some error")' will call env.abort api.
```
      call $~lib/builtins/abort
```
In iwasm, env.abort is implemented here.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/a5188f5574a0f0aed7d178ca3774ff8657f4ac95/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c#L978-L985
But in [AssemblyScript](https://github.com/AssemblyScript/assemblyscript/blob/f1282f3bd0376386675486d32ad8b01114ef75fe/std/assembly/builtins.ts#L2142-L2149), the signature of abort is difference with iwasm runtime.
```
// @ts-ignore: decorator
@external("env", "abort")
declare function abort(
  message?: string | null,
  fileName?: string | null,
  lineNumber?: u32,
  columnNumber?: u32
): void;
```
I am not sure which cause this error or is my understanding wrong, thanks.

* ARCH: mac RELEASE_X86_64 x86_64 
* Run mode: Intercepter
* [abort_test.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5910270/abort_test.wasm.zip)

wenyongh, 2021-02-04:
Hi, it seems that the env.abort native api is required by both libc-builtin and AssemblyScript, currently it is registered with signature "(i)" to support libc-builtin by default. Normally you can also register your own env.abort native api to support AssemblyScript, but in runtime now, the registered node is appended into the end of list, so runtime finds libc-builtin's api firstly. We might change runtime behavior to insert it to the head of list so as to support finding developer's registered api firstly. I uploaded a patch your reference, could you please test it? Thanks.
[register_native.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5923854/register_native.zip)

JavanZhu, 2021-02-04:
Hi wenyongh, thanks for your reply, and I'll test later.

JavanZhu, 2021-02-04:
@wenyongh Thanks a lot, I'll try it later and close this issue later.

closed 2021-02-04 by JavanZhu