wasm-micro-runtime/wasm-micro-runtime #512: What is the best way to run multiple modules at the same time

oubotong, 2021-02-05:
I have some questions regarding the multi-module scenario of WAMR. In our case, we are trying to spawn multiple WASM modules on the device(for different clients) and we have two intuitive potential options to achieve this(let's assume we run WAMR with Zephyr):

1. We create multiple threads in Zephyr to launch multiple WAMR instances and each of them is responsible for load and execute different WASM modules. 

2. Is it possible to create multiple threads in WAMR to load different WASM modules and execute them at the same time? If yes, then is WAMR implementation thread-safe? Do I need to add my own thread handler in WAMR?

Thanks in advance!

xujuntwt95329, 2021-02-05:
Hi, seems you want to run multiple application on your device ?

You can create multiple threads through platform APIs such as `os_thread_create`, then you can load, instantiate and execute the module in there own threads.

BTW, WAMR has provided an application framework, it's called app-manager, it has already provided the functionality of installing/uninstalling/running multiple wasm modules, and we also developed some libraries to support timers, sensors and cross-application communications, please refer to [application framework](https://github.com/bytecodealliance/wasm-micro-runtime#application-framework) for more details.

oubotong, 2021-02-05:
If I create multiple threads for WAMR, then I think that will introduce lots of memory overhead(each module would require an individual WAMR instance).  In the application framework, are the pub app and sub app running simultaneously on the SoC within a single WAMR? or within different WAMR.

xujuntwt95329, 2021-02-05:
> If I create multiple threads for WAMR, then I think that will introduce lots of memory overhead(each module would require an individual WAMR instance

There is only a single WAMR, each module will require an individual module instance rather than the whole runtime.

oubotong, 2021-02-05:
> There is only a single WAMR, each module will require an individual module instance rather than the whole runtime.
Right, so according to my understanding, there will be two wasm module instances to be initiated in WAMR. Then how does WAMR handle the context switch between these two modules if they are running at the same time?

xujuntwt95329, 2021-02-06:
WAMR can't do the context switch, the `os_thread_create` API is just a wrapper of the OS's thread interface. The context switch is handled by the OS, there may be different policies on different platforms. 

WAMR's application framework introduced an event-driven programming model, the user application just register some event handlers and waiting for the events.

closed 2021-05-27 by wenyongh