wasm-micro-runtime/wasm-micro-runtime #513: On Android platform，wasm-micro-runtime initialized but not destroied，multiple threads creates will be failed for the second time.

leehee7, 2021-02-05:
On Android platform，wasm-micro-runtime is initialized but not destroied，the first time multiple threads are successfully created, however, the same operation will be failed for the second time.

The bug is:
core\iwasm\interpreter\wasm_runtime.c -> memories_deinstantiate
1. os_mutex_destroy(&memories[0]->mem_lock); -> os_mutex_destroy(&memories[i]->mem_lock);
2. and os_mutex_destroy(&memories[i]->mem_lock) shoud be done when ref_count = 0

please fix it.

wenyongh, 2021-02-07:
@leehee7 Thanks a lot for reporting the issue! We have fixed it in patch #514 , could you please try again? Thanks.

xingkaiyu, 2021-02-09:
@wenyongh  @leehee7  I have tryed it using 2021-02-07 code and it is ok now. thanks~

closed 2021-02-09 by wenyongh