wasm-micro-runtime/wasm-micro-runtime #516: The main thread executes pthread_join, the correct result cannot be obtained

leehee7, 2021-02-07:
The child thread finishes earlier than the main thread，the main thread executes pthread_join, the correct result cannot be obtained.
___________________________________________________________________________________
if (0 == pthread_create(&threadNo, NULL, ThreadTransChar, paramChar)) {
    MyDelay(delayTime);  // call native function doing delay, the child thread will finish earlier than the main thread
    void *pResult;
    int joinResult = pthread_join(threadNo, &pResult);
    char *pResultChar = (char *)pResult;
    LOG(TAG, "Test DATA_TPYE_CHAR joinResult is %d", joinResult);
    LOG(TAG, "Test DATA_TPYE_CHAR pResult is %p, result is %s", pResultChar, pResultChar);
    free(pResultChar);
}
___________________________________________________________________________________

wenyongh, 2021-02-08:
Hi, it should be normal behavior, similar to Linux pthread API's behavior. The child thread can end earlier than main thread. If you want, maybe you should add some synchronization mechanism between main thread and child thread, e.g. the child thread should wait until the main thread's MyDelay() operation finishes.

xingkaiyu, 2021-02-08:
Hi wenyongh, I tested the pthread_join behavior in android native, but get the different result.
my main thread method is like this:
(param is_delay is using to control whether the pthread_join happans before or after the end of the child thread)
extern "C" JNIEXPORT jint JNICALL
Java_com_example_myapplication_MainActivity_testPthreadNative(JNIEnv *env, jobject thiz, jboolean is_delay) {
    pthread_t threadNo;

    if (0 == pthread_create(&threadNo, NULL, ThreadTest, NULL)) {
        if (is_delay) {
            usleep(1000 * 500);
        }
        void *result;
        int joinResult = pthread_join(threadNo, &result);
        if (0 == joinResult) {
            int *temp = (int *)result;
            return *temp;
        } else {
            return -1;
        }
    }

    return -1;
}

and then I have two ways to write my ThreadTest function:

1- return a local int address:
void *ThreadTest(void *)
{
    usleep(1000 * 100);
    int a = 1;
    return &a;
}
and then I will always got crash(Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR)), no matter I set is_delay true or false.

2- return a int address by new int:
void *ThreadTest(void *)
{
    usleep(1000 * 100);
    int *a = new int(1);
    return a;
}
and then I will always got the right result 1, no matter I set is_delay true or false.

But when I write same code in wasm, no matter what kind of address I returned in ThreadTest(local or new), I will got the right result when main thread not delay and got 0 when main thread delay.

So I don't know whether the android behavior or the wamr is normal?

wenyongh, 2021-02-09:
Hi, @xingkaiyu, in the first way, you return the address of local variable to outside, and access it in the main thread, so it might run into crash. For wasm, it is a sandbox environment, the app's address is the relative offset the linear memory with range [0, linear_mem_size), the local variable address might also be inside the linear memory space, so you might got some result value. And if the address is outside the linear memory space, accessing to the address will get "out of bounds memory access" exception.

We have submitted the patch #522 to enable pthread joining the exited thread, could you please try it again?

xingkaiyu, 2021-02-19:
Hi, @wenyongh , the result is changed after your patch. But I still think it is a little strange.
I write function as before in wasm:
void TestJoinResult(bool isDelay)
{
    pthread_t threadNo;
    if (0 == pthread_create(&threadNo, NULL, ThreadJoinResult, NULL)) {
        if (isDelay) {
            MyDelayUS(1000 * 500);
        }
        void *result;
        int joinResult = pthread_join(threadNo, &result);
        if (0 == joinResult) {
            int *temp = (int *)result;
            LOGI("TestJoinResult pthread_join result %d", *temp);
        } else {
            LOGI("TestJoinResult pthread_join error");
        }
    }
}
MyDelayUS is a native function which use usleep to delay by us. LOGI is a log function help to check the result.

when I write ThreadJoinResult without new like below, I will always got 0 no matter I set delay true or false:
void *ThreadJoinResult(void *)
{
    MyDelayUS(1000 * 100);
    int a = 1;
    return &a;
}

when I write ThreadJoinResult with new like below, I will got 0 when I set delay true and 1 when I set delay false. (And if I set delay false first and got the result 1, I will then got 1 when I set delay true. I don't know why...):
void *ThreadJoinResult(void *)
{
    MyDelayUS(1000 * 100);
    int *a = new int(1);
    return a;
}

xingkaiyu, 2021-02-20:
@wenyongh  maybe my code haven't update...  I will try agin and get the new result. sorry...

xingkaiyu, 2021-02-20:
@wenyongh   @leehee7   I used new code today and checked the result is:
1- when use new or malloc to return, I always get the right result.
2- when use local var to return, I result is random. Some time is 0 and some time is other number.
I think it is better now and The issue can be closed. Thanks~

leehee7, 2021-02-20:
thanks，the issue have resolved

closed 2021-02-20 by leehee7