wasm-micro-runtime/wasm-micro-runtime #518: In wasi-12.0,  using double type in a child thread, i can not get the correct result on Android.

leehee7, 2021-02-07:
In wasi-12.0,  using double type in a child thread, i can not get the correct results on Android.
for example:
###1###
when i use vsprintf or sprintf to format string, always get 0.000 for double.
char *pTest = “32423.54”;
sprintf(msg, "ThreadDouble test sprintf is %f", *(pTest)); // the result always is 0.000

###2###
When the following codes is executed，wasm runtime will be abnormal some time in child thread，reporting "out of bounds memory access", unless i reLoad the wasm runtime.
void LogInfoHere(const char *tag, const char *fmt, ...)
{
    char msg[1024];
    va_list va;
    va_start(va, fmt);
    vsprintf(msg, fmt, va);
    va_end(va);
    LogInfoNative(tag, msg);
}
LogInfoHere(TAG, "ThreadDouble test is %f", 32423.54);

wenyongh, 2021-02-08:
Hi, for the issue 1, it should be normal as the pTest variable type is `char *` but not `double *`, *(pThread) is char '3', the double value converted seems to be 0.000.

For the issue 2, how do you build and run the wasm app? Is there more complete example to reproduce it? Thanks.

xingkaiyu, 2021-02-09:
Hi @wenyongh , the description before is not right. The problem we face is like this:
I have write same code both in android native and wasm(using wasi-sdk 12, run in android)
void *ThreadDouble(void *in)
{
    double *inDouble = (double *)in;
    LOGI("inDouble is %f", *inDouble);

    double *a = new double(*inDouble * 2);
    LOGI("a is %p, *a is %f", a, *a);  // line 1
    return a;
}

void TestPthreadDouble()
{
    pthread_t threadNo;
    double param = 12.7;
    LOGI("pthread_join param %f", param);
    if (0 == pthread_create(&threadNo, NULL, ThreadDouble, &param)) {
        void *result;
        int joinResult = pthread_join(threadNo, &result);
        if (0 == joinResult) {
            double *temp = (double *)result;
            LOGI("pthread_join result %f", *temp);
        } else {
            LOGI("pthread_join failed %f", joinResult);
        }
    }
}

of course the function LOGI is not the same, in android LOGI is like this:
#define LOGI(...) __android_log_print(ANDROID_LOG_INFO, LOG_TAG, __VA_ARGS__)

in wasm LOGI is like this:
void LOGI(const char *fmt, ...)
{
    char msg[1024];
    va_list va;
    va_start(va, fmt);
    vsprintf(msg, fmt, va);
    va_end(va);

    LogInfoNative(TAG, msg);
}
LogInfoNative is a function I export to wasm in android native:
void LogInfoNative(wasm_exec_env_t exec_env, const char *tag, const char *msg)
{
    LogI(tag, msg);
}
and logI is:
void LogI(const char *tag, const char *fmt)
{
    __android_log_write(ANDROID_LOG_INFO, tag, fmt);
}

And then, When I run it in android, I will get the log like this:
I/native-lib: pthread_join param 12.700000
I/native-lib: inDouble is 12.700000
I/native-lib: a is 0x78db04b670, *a is 25.400000
I/native-lib: pthread_join result 25.400000

But when I run it in wasm, I have two problems:
1- If I delete "line 1" in ThreadDouble, the log is like this:
I/TestPthread: pthread_join param 12.700000
I/TestPthread: inDouble is 0.000000
I/TestPthread: pthread_join result 25.400000
double value is wrong when I use string formart in native, but the double result return to main thread is ok.

2- If I do not delete "line 1" in ThreadDouble, the log is like this:
I/TestPthread: pthread_join param 12.700000
I/TestPthread: inDouble is 0.000000
"out of bounds memory access" error
wamr runtime went abnormal, and when I try to call the same function again, I always got "out of bounds memory access" error and the wasm function will not enter any more unless I reload the wasm instance.

for problem 2, I did some other test. Some times, the second time error will be like "function need 2 param but pass 0" error (first time error always "out of bounds memory access" error). But sorry I haven't record the code for this scenes

wenyongh, 2021-02-10:
@xingkaiyu seems that the issue is caused by that auxiliary stack allocated for new thread isn't 16-byte aligned, we have fixed it in #524, could you please try again? Thanks.

xingkaiyu, 2021-02-20:
@wenyongh  I think it ok now. Thanks~

closed 2021-02-23 by wenyongh