wasm-micro-runtime/wasm-micro-runtime #525: Can't compile WASI with Windows

Paril, 2021-02-10:
Been trying to get WASI working with Windows, no luck so far though.

With the following configuration:
`cmake .. -DWAMR_BUILD_LIBC_BUILTIN=0 -DWAMR_BUILD_LIBC_WASI=1`

many errors are spat out, some related to pthreads' types missing (pthreads was not enabled) but the majority pointing to issues related to "alignof".

```wasm-micro-runtime\core\iwasm\libraries\libc-wasi\sandboxed-system-primitives\include\wasmtime_ssp.h(32
,16): error C2143: syntax error: missing ')' before 'alignof' [wasm-micro-runtime\product-mini\platform
s\windows\build\vmlib.vcxproj]``

It seems to be because these files are compiled with MSVC's (fairly bad) toolkit for C.

xujuntwt95329, 2021-02-10:
Hi, currently we haven't supported WASI on Windows, the current implementation relies on many POSIX APIs.

Here is an issue about WASI implementation on Windows which uses `uvwasi`: #499.

Paril, 2021-02-10:
Is the issue just that you need POSIX wrappers (and preprocessor branches to support Legacy MSVC)? I can probably implement some if that's the major blocking point.

xujuntwt95329, 2021-02-10:
Actually I haven't evaluated how much work need to be done for supporting windows, so I'm not sure whether there are any other gaps.

Personally speaking, I think `uvwasi` should be a better choice for Windows since it has been working well in nodejs.

Seems [this fork](https://github.com/bxkai/wasm-micro-runtime/commit/fc8ba7f3039b5c58c3f77920b517473f0751531f) has done a lot of works, maybe you can have a try.

Paril, 2021-02-11:
Interesting; I'll take a look.

How does the builtin-libc work, by comparison? I've been trying to wrap my head around varargs, for instance. Clang can compile functions with varargs - and can even call imported functions that expect varargs passed to them - but, on the native side, I don't know what the type of "data" it is that it's expecting. Is this documented anywhere? In the builtin libc, it appears to expect the varargs to be a string (it uses the $ parameter type), but I'm wondering if this is standardized anywhere.

closed 2021-02-23 by wenyongh

xujuntwt95329, 2021-02-23:
The `va_args` is not a string, it's a pointer.  Let's take `printf` for example, it is registered as:
``` C
REG_NATIVE_FUNC(printf, "($*)i"),
```

The first parameter is defined as `$`(string), the second param is `*`(pointer). And the prototype of the wrapper is:
``` C
static int
printf_wrapper(wasm_exec_env_t exec_env,
               const char * format, _va_list va_args)
```

The `$` is for the `format` parameter rather than `va_args`.