wasm-micro-runtime/wasm-micro-runtime #526: Can't import functions without a built-in module name

Paril, 2021-02-15:
If the `import_module` attribute is set to something other than "env", it is impossible to import it it seems. I've registered the natives with `wasm_runtime_register_natives` with my specified module name, but since it doesn't consider it "built in" it never checks/links them to imports.

I assume this is mainly for multi-module stuff, but, it would be nice to allow this to work on its own.

wenyongh, 2021-02-16:
Hi, what the import module name should be depends on the wasm file generated, in which the import function has attribute import module name and import field name. The module_name passed to wasm_runtime_register_natives should match the import module name of the import functions. You may use wasm-objdump tool of wabt to dump the details. Not sure how to let toolchain e.g. wasi-sdk or emcc generate specified import module name for wasm file.

And note that the wasm_runtime_register_natives should be called before the wasm file is loaded, or the loading process can not find the symbol and link it.

Paril, 2021-02-16:
Yeah, that's all well and good - it's just that the functions aren't actually linked in unless the module name is a special, built-in one.

I've been trying to use a custom module name, but it fails to link them during registration. Peeking through the code shows that it will only actually link them in if it is a special name, such as "env", *unless* multi-modules are enabled.

wenyongh, 2021-02-18:
Hi, thanks for finding the issue! The loader's behavior is really not as expected, I uploaded a patch to fix it, could you please try again? We will test more cases and merge it soon.
[import_non_builtin_module.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6001895/import_non_builtin_module.zip)

Paril, 2021-02-19:
Yep, that works!

closed 2021-02-23 by wenyongh