wasm-micro-runtime/wasm-micro-runtime #528: HW trap on SGX

minyeong97, 2021-02-18:
Hi. First of all, thank you for your dedication on this project. Really enjoying wasm-micro-runtime!

I have some questions regarding the HW traps on SGX. Far as I know, HW trap bound checking is disabled in SGX. Is this due to a technical reason? or is it not implemented yet?

Based on my research, it is possible to mark pages mapped to SGX EPCs as PROT_NONE (please correct me if I am wrong). If this is possible, I think SGX wasm-micro-runtime can take advantage of hardware memory protection as same as non-SGX runtimes.

Thanks in advance!

wenyongh, 2021-02-18:
Hi, thanks for supporting wasm-micro-runtime! For SGX, there are really some technical issues to support boundary check with HW trap: (1) we wrap mmap by calling sgx_alloc_rsrv_mem() function, which cannot allocate memory of 8GB size with PROT_NONE flag set. Another way might be calling host's mmap with ocall mechanism to mmap 8GB PROT_NONE memory, and then calling sgx_alloc_rsrv_mem() to allocate Enclave's memory with fixed address (= address of host's mmap). Not sure whether this works, and even if it works, there is potential security issue as it depends on ocall. (2) the SGX exception handler is different from posix signal handler, from the latter, we can learn which address causes the signal and check whether the address is in the range of guard pages to confirm whether it should be a memory boundary check failure. But in SGX exception handler, it only provides sgx_cpu_context_t structure with CPU registers value, it is hard to learn whether the exception is caused by accessing guard pages.

The HW trap boundary check mechanism really improves performance a lot, we might investigate more in the future.

minyeong97, 2021-02-19:
Thank you for your reply.

Can you elaborate on why we can't allocate 8GB in SGX with PROT_NONE? After I reading your comment, I did some research to figure out why that is the case. I found out that there is some kind of PROT_NONE setting for memory allocated inside SGX from [intel's sgx sdk](https://github.com/intel/linux-sgx/blob/bb3d1a5a302511954fcd1b20df4466554e129df1/sdk/tmm_rsrv/sgx_rsrv_mem.cpp#L327). The code shows that it is possible to update EPCM for memory protection. I can be wrong, due to the fact that I didn't and can't run Intel SGX on my machine.

Thanks again!

wenyongh, 2021-02-20:
There are some APIs provided by SGX SDK, normally we can use sgx_alloc_rsrv_mem() or sgx_alloc_rsrv_mem_ex() to allocate memory, and use sgx_tprotect_rsrv_mem() to set the memory protection attributes:
```C
void * sgx_alloc_rsrv_mem(size_t length);
void * sgx_alloc_rsrv_mem_ex(void * desired_addr, size_t length);
sgx_status_t sgx_tprotect_rsrv_mem(void *addr, size_t length, int prot);
```
The sgx_alloc_rsrv_mem()/sgx_alloc_rsrv_mem_ex() doesn't provide argument to set the memory protection attributes like mmap(), the memory allocated is RW, and the size must be no larger than the reserved memory size set in Enclave.config.xml, and it seems that the reserved memory is pre-allocated from physical memory. In other words, to allocate 8GB memory from reserved memory, the total machine's physical memory and the reserved memory configured in Enclave.config.xml must be larger than 8GB. That might not be acceptable in many situations. Also it is very slow to launch the enclave and allocate the 8GB memory, which might also be unacceptable.

Another issue is that function sgx_tprotect_rsrv_mem() is only "workable on the Intel(R) SGX 2.0 enabled platform. For
Intel(R) SGX 1.0 this function doesn't actually change any permission but perform a sanity check." So setting memory page to PROT_NONE might not take effect.

closed 2021-06-16 by wenyongh