wasm-micro-runtime/wasm-micro-runtime #532: Log function passthrough support

Paril, 2021-02-19:
I'm working in an environment where stdout isn't immediately available to me, but would like to trap perf, consumption and stack trace data - bh_log only seems to directly pump to stdout. Would there be any way of adding a log function that the final strings could be sent through to? Alternatively, having a way to iterate performance/mem stats/stack traces without printing would be handy.

lum1n0us, 2021-02-23:
Sure.  `bh_log()` actually depends on `os_printf()` in `core/shared/platform/PLATFORM_NAME/platform_init.c`. in other words, `os_printf()` will control the target output file. feel free to modify the implementation to send log information to where you need.

for example, on Linux platform, using `int vfprintf(FILE *stream, const char *format, va_list ap);` instead of `int vprintf(const char *format, va_list ap)` is an option.

Paril, 2021-02-23:
Yeah I'm just wondering if it's something that can be done at library level rather than me hacking around it. I need to essentially send a callback function to it, because I'm loading this through a DLL in an EXE so I don't have access to stdout, and would rather it print to the console of the game. Would a patch of that be useful?

lum1n0us, 2021-02-28:
I am not sure that I fully understand the meaning of *library level* support you mentioned. WAMR is kind of all-in-one for now.

lum1n0us, 2021-03-01:
$ # implement a function likes: int my_printf(const char *format, va_list ap) { //output to a file }
$ cmake -DBH_VPRINTF=my_printf ..
$ make

[9804e21.diff.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6059685/9804e21.diff.zip)

Paril, 2021-03-01:
@lum1n0us Sorry, I'll clarify a little bit.

I'm using WAMR as a component of a larger engine. The engine already has its own logging facilities, and stdout/in/err are useless to it (especially on Windows, where it's 100% useless: there is no stdout/in/err at all, so the messages get voided entirely). It is cross-platform, and it's a bit silly for us to edit every single platform to support a new log function - and even if we did, there's no "global" function I can call to hook back into our engines' logging facility. There's no global stream I can access via WAMR. It's all local to the game's engine.

What I need is the ability to have the strings from the logger in WAMR sent back to us for logging. Think OpenGL's debug logging facility, inwhich you register a callback function that gets messages sent back to you.

wenyongh, 2021-03-02:
@Paril as all log outputs including bh_log() API, call stack dump, memory/perf profiling and other logs, eventually call the os_vprintf funciton to print the string (except library wasi implementation might cause writev/pwritev instead), and it is relatively complex and might be big modification to register the log callback function, we prefer to let developer define the BH_VPRINTF macro to his self-defined vprintf API to output the log he wants. Could you please patch @lum1n0us uploaded? After patching it, you can implement your vprintf like function, and define the macro BH_VPRINTF in you makefile, then the os_printf and os_vprintf will call your vprintf funciton to output the log instead calling libc vprintf in platform linux and darwin, e.g.:
```C
int my_vprintf(const char *format, va_list ap) { ... }
CMakeLists.txt: add_definitions(-DBH_VPRINTF=my_printf)
```
Could you please try whether it works for you? Thanks.

Paril, 2021-03-02:
I suppose since it's statically linked, I could have that function implemented in the exe and the vmlib could just leave it undefined so that I'm forced to include it. I'll give it a shot.

closed 2021-03-16 by wenyongh