wasm-micro-runtime/wasm-micro-runtime #535: Param transfer problem when using pthread

xingkaiyu, 2021-02-20:
Hi, I got a problem when using pthread in wamr (platform is android; using 2021-02-20 wamr code; using wasi-sdk-12.0)
My code is like this: (TestTransParamParallel is the export function)

struct TransParamParallelStruct
{
    bool boolParam;
    int intParam;
};

void *ThreadTransParamParallel(void *arg)
{
    TransParamParallelStruct transParamParallelStruct = *(TransParamParallelStruct *)arg;
    LOGI("ThreadTransParamParallel threadNo is %d intParam is %d boolParam is %d", 
        pthread_self(), transParamParallelStruct.intParam, transParamParallelStruct.boolParam);
    return NULL;
}

void TestTransParamParallel()
{
    LOGI("TestTransParamParallel in");
    std::vector<pthread_t> threadNos;

    std::vector<TransParamParallelStruct> transParamParallelStructs;
    for (int i = 0; i < 4; i++) {
        TransParamParallelStruct transParamParallelStruct;
        transParamParallelStruct.boolParam = true;
        transParamParallelStruct.intParam = i;
        transParamParallelStructs.push_back(transParamParallelStruct);
    }
    for (TransParamParallelStruct transParamParallelStruct : transParamParallelStructs) {
        pthread_t threadNo;
        if (0 == pthread_create(&threadNo, NULL, ThreadTransParamParallel, &transParamParallelStruct)) {
            threadNos.push_back(threadNo);
        }
    }

    for (pthread_t threadNo : threadNos) {
        pthread_join(threadNo, NULL);
    }

    LOGI("TestTransParamParallel out");
}

And the problem is when I call the function the int param is not always 0 1 2 3 in four sub thread. For example, I may get the log like this:
ThreadTransParamParallel threadNo is 9 intParam is 0 boolParam is 1
ThreadTransParamParallel threadNo is 10 intParam is 2 boolParam is 1
ThreadTransParamParallel threadNo is 11 intParam is 2 boolParam is 1
ThreadTransParamParallel threadNo is 12 intParam is 3 boolParam is 1

So I think it may has some problem in wamr. Please check~

wenyongh, 2021-02-22:
Hi, this is because that the argument transParamParallelStruct passed to the thread callback function is a local variable, it is temporarily variable and inside the stack area, each thread callback's argument points to the same address on stack, so their value might be same. We can get the similar results under Linux with g++ compiler. So please had better pass the argument to the thread callback function with singleton/standalone address, e.g. a newly allocated memory:
```C
void *ThreadTransParamParallel(void *arg)
{
    access arg;
    free arg;
    ...
}

void TestTransParamParallel()
{
    ...
    for (TransParamParallelStruct transParamParallelStruct : transParamParallelStructs) {
        pthread_t threadNo;
        TransParamParallelStruct *transParamParallelStruct_new = new TransParamParallelStruct[1];

        *transParamParallelStruct_new = transParamParallelStruct;
        if (0 == pthread_create(&threadNo, NULL, ThreadTransParamParallel, transParamParallelStruct_new)) {
            threadNos.push_back(threadNo);
        }
        else {
            // free memory if pthread_create failed
            delete []transParamParallelStruct_new;
        }
    }
    ...
}
```
Could you please try again in such way?

xingkaiyu, 2021-02-22:
@wenyongh  
I have tryed it with var create by new. It is ok now. Thanks~
But is it mean if I want to get right param in sub thread, I must always use new or malloc in main thread (not only this Scene), and I need delete the param in sub thread when pthread create success but delete param in main when create failed?

wenyongh, 2021-02-22:
Hi, it depends, for the buffer, e.g. structure or string, had better do it in such way, or maintain the buffer by yourself. For immediate number, e.g. int x, maybe we can pass it as `(void*)(uintptr_t)x`, and then in the callback function, get it with `x = (int)(uintptr_t)arg`.

closed 2021-02-23 by wenyongh