wasm-micro-runtime/wasm-micro-runtime #536: iwasm exception: mprotect failing while initializing the heap

djrenren, 2021-02-20:
**Platform:** Linux x86_64 (Archlinux)
**Product:** iwasm

**Command that fails:**
```
./iwasm hello_world.wasm
```

**Output:**
```
Exception: set protected page failed
```

**Test file**: The default rust hello world program compiled with `cargo wasi`. [hi.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6016292/hi.wasm.zip) (zipped to make Github happy)
**Build options for iwasm:** All default

**Investigation so far:**

I did some further digging and added a line to [aot_runtime.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/aot/aot_runtime.c#L1070) to print the errno:
```diff
+ printf("ERROR: %d, %s\n", errno, strerror(errno));
  aot_set_exception(module_inst, "set protected page failed");
```

and got: `ERROR: 12, Cannot allocate memory`

As you might expect, if I build iwasm with `cmake .. -DWAMR_DISABLE_HW_BOUND_CHECK=1` then I no longer get this error and the program correctly prints:

```
Hello, World!
```

I'm not very up on how mprotect works so I'm not sure how to fix the bug. Any thoughts?

wenyongh, 2021-02-22:
Hi, not sure what causes the problem, from the log, the mprotect function is used to set the attributes of the native stack guard pages, and the stack_min_addr is obtained by os_thread_get_stack_boundary() in file core/shared/platform/common/posix/posix_thread.c:
```C
#ifdef __linux__
    if (pthread_getattr_np(self, &attr) == 0) {
        pthread_attr_getstack(&attr, (void**)&addr, &stack_size);
        pthread_attr_getguardsize(&attr, &guard_size);
        ...
```
Could you please check whether these codes are correctly executed? If yes, from the errno str "Cannot allocate memory",  it seems that there are not enough available memory pages to allocate, not sure whether the default max native stack size is too large, which is set as 8M by default in core/config.h with macro APP_THREAD_STACK_SIZE_MAX, could you please manually set it in CMakeLists.txt, e.g.:
```cmake
add_definitions(-DAPP_THREAD_STACK_SIZE_MAX=1047586)
```
Thanks.

djrenren, 2021-02-22:
I added the following lines:

```diff
         pthread_attr_getguardsize(&attr, &guard_size);
+        printf("guard_size: %d\n", APP_THREAD_STACK_SIZE_DEFAULT);
+        printf("STACK SIZE: %zu\n", stack_size);
+        printf("MAX STACK SIZE: %zu\n", max_stack_size);
+        printf("STACK SIZE DEFAULT: %d\n", APP_THREAD_STACK_SIZE_DEFAULT);
```

And got the following output:
```
guard_size: 32768
STACK SIZE: 8384512
MAX STACK SIZE: 8388608
STACK SIZE DEFAULT: 32768
```

I also added a log for the arguments to mprotect:
```
            printf("os_mprotect(%p, %d, %d)\n", stack_min_addr, page_size * guard_page_count, MMAP_PROT_NONE);
```

that yielded:
```
os_mprotect(0x7ffca6846000, 12288, 0)
```

There's plenty of available memory so that shouldn't be what's causing it...

wenyongh, 2021-02-23:
Hi, @djrenren, the stack page mapping may be lazily grown on some OS, which means that not all the pages are pre-mapped (or pre-allocated) unless they are actually used, normally they are mapped one by one from current page to the bottom page gotten from function pthread_attr_getstack(.., &addr, &stack_size). For the HW trap for native stack overflow check, we get the bottom page address, and touch each page one by one to ensure that all the pages are pre-allocated. So one issue may be the stack size is too large, another issue may be touching pages doesn't take effect. For the latter, could you please check whether the code to write the page in aot_runtime.c function touch_pages() is optimized by compiler:
```C
#if defined(__GNUC__)
__attribute__((no_sanitize_address)) static uint32   => no_sanitize_address inform the compiler that it should not instrument memory accesses in the function when compiling with the -fsanitize=address option 
#else
static uint32
#endif
touch_pages(uint8 *stack_min_addr, uint32 page_size)
{
    ...
        *touch_addr = 0;                   => write the page to ensure it is mapped, should not be optimized by compiler
        sum += *touch_addr;
    ...
}
```
Could you please confirm that? If it is not the root cause, could you please set APP_THREAD_STACK_SIZE_MAX to smaller size e.g. 1M/512KB and try again? Thanks.

djrenren, 2021-02-23:
Ah! Sure enough the problem only replicates on -O2 and -O3. Here's a disassembly of touch_pages on O2:

```
   0x000000000000f3c0 <+0>:     push   %rbp
   0x000000000000f3c1 <+1>:     mov    %rsp,%rbp
   0x000000000000f3c4 <+4>:     sub    $0x10,%rsp
   0x000000000000f3c8 <+8>:     mov    %fs:0x28,%rax
   0x000000000000f3d1 <+17>:    mov    %rax,-0x8(%rbp)
   0x000000000000f3d5 <+21>:    mov    %esi,%eax
   0x000000000000f3d7 <+23>:    mov    %esi,%esi
   0x000000000000f3d9 <+25>:    add    %rsi,%rdi
   0x000000000000f3dc <+28>:    shr    %eax
   0x000000000000f3de <+30>:    lea    0x17(%rax),%rdx
   0x000000000000f3e2 <+34>:    and    $0xfffffff0,%edx
   0x000000000000f3e5 <+37>:    sub    %rdx,%rsp
   0x000000000000f3e8 <+40>:    lea    0xf(%rsp),%rax
   0x000000000000f3ed <+45>:    and    $0xfffffffffffffff0,%rax
   0x000000000000f3f1 <+49>:    cmp    %rdi,%rax
   0x000000000000f3f4 <+52>:    jb     0xf417 <touch_pages+87>
   0x000000000000f3f6 <+54>:    nopw   %cs:0x0(%rax,%rax,1)
   0x000000000000f400 <+64>:    sub    %rdx,%rsp
   0x000000000000f403 <+67>:    movb   $0x0,(%rax)
   0x000000000000f406 <+70>:    movzbl (%rax),%eax
   0x000000000000f409 <+73>:    lea    0xf(%rsp),%rax
   0x000000000000f40e <+78>:    and    $0xfffffffffffffff0,%rax
   0x000000000000f412 <+82>:    cmp    %rdi,%rax
   0x000000000000f415 <+85>:    jae    0xf400 <touch_pages+64>
   0x000000000000f417 <+87>:    mov    -0x8(%rbp),%rax
   0x000000000000f41b <+91>:    sub    %fs:0x28,%rax
   0x000000000000f424 <+100>:   jne    0xf428 <touch_pages+104>
   0x000000000000f426 <+102>:   leaveq 
   0x000000000000f427 <+103>:   retq   
   0x000000000000f428 <+104>:   callq  0x61b0 <__stack_chk_fail@plt>
   ```
   
   vs with 01
   
   ```
      0x000000000000ec02 <+0>:     push   %rbp
   0x000000000000ec03 <+1>:     mov    %rsp,%rbp
   0x000000000000ec06 <+4>:     sub    $0x10,%rsp
   0x000000000000ec0a <+8>:     mov    %rdi,%r9
   0x000000000000ec0d <+11>:    mov    %esi,%r8d
   0x000000000000ec10 <+14>:    mov    %fs:0x28,%rax
   0x000000000000ec19 <+23>:    mov    %rax,-0x8(%rbp)
   0x000000000000ec1d <+27>:    xor    %eax,%eax
   0x000000000000ec1f <+29>:    shr    %esi
   0x000000000000ec21 <+31>:    mov    %esi,%esi
   0x000000000000ec23 <+33>:    lea    0x17(%rsi),%rax
   0x000000000000ec27 <+37>:    and    $0xfffffff0,%eax
   0x000000000000ec2a <+40>:    sub    %rax,%rsp
   0x000000000000ec2d <+43>:    lea    0xf(%rsp),%rax
   0x000000000000ec32 <+48>:    and    $0xfffffffffffffff0,%rax
   0x000000000000ec36 <+52>:    mov    %rax,%rdx
   0x000000000000ec39 <+55>:    mov    %r8d,%r8d
   0x000000000000ec3c <+58>:    add    %r8,%rdi
   0x000000000000ec3f <+61>:    mov    $0x0,%ecx
   0x000000000000ec44 <+66>:    add    $0x17,%rsi
   0x000000000000ec48 <+70>:    and    $0xfffffff0,%esi
   0x000000000000ec4b <+73>:    cmp    %rdi,%rax
   0x000000000000ec4e <+76>:    jb     0xec6c <touch_pages+106>
   0x000000000000ec50 <+78>:    movb   $0x0,(%rdx)
   0x000000000000ec53 <+81>:    movzbl (%rdx),%eax
   0x000000000000ec56 <+84>:    add    %eax,%ecx
   0x000000000000ec58 <+86>:    sub    %rsi,%rsp
   0x000000000000ec5b <+89>:    lea    0xf(%rsp),%rax
   0x000000000000ec60 <+94>:    and    $0xfffffffffffffff0,%rax
   0x000000000000ec64 <+98>:    mov    %rax,%rdx
   0x000000000000ec67 <+101>:   cmp    %rdi,%rax
   0x000000000000ec6a <+104>:   jae    0xec50 <touch_pages+78>
   0x000000000000ec6c <+106>:   add    -0x1(%r9,%r8,1),%cl
   0x000000000000ec71 <+111>:   movzbl %cl,%eax
   0x000000000000ec74 <+114>:   mov    -0x8(%rbp),%rcx
   0x000000000000ec78 <+118>:   sub    %fs:0x28,%rcx
   0x000000000000ec81 <+127>:   jne    0xec85 <touch_pages+131>
   0x000000000000ec83 <+129>:   leaveq 
   0x000000000000ec84 <+130>:   retq   
   0x000000000000ec85 <+131>:   callq  0x61b0 <__stack_chk_fail@plt>

closed 2021-06-09 by wenyongh