wasm-micro-runtime/wasm-micro-runtime #537: Passing NULL from WASM to Native

Paril, 2021-02-21:
I ran into this issue in the project I'm working on, where at times a native function with a pointer signature like `(*)` may be given NULL. On the native side, however, this value is offset by the WASM heap address (it's the same value as `wasm_runtime_addr_app_to_native(module_inst, 0)`. Is this an oversight or something I'm unaware of? Is it ever possible for this value to come in as a non-NULL value? Should WAMR just treat 0 in pointer sigs as 0 instead of translating them?

wenyongh, 2021-02-22:
Hi, do you mean that the wasm app passes argument NULL pointer to native API, and the NULL pointer is converted to 0 by the WASM toolchain, so the native API gets 0 as offset to linear memory, and in WAMR, if the native API's argument signature is configured with `*` as pointer, function `wasm_runtime_addr_app_to_native` converts it into the native address of linear memory's begin address, but not NULL?

Yes, we might treat 0 as NULL to return in function wasm_runtime_addr_app_to_native, but it might not be a common case, for C it might be OK to convert 0 to NULL, but for other languages, e.g. Rust/AssemblyScript, it might be invalid. The WASM spec defines that the linear memory address range is 0 to linear_mem_size - 1, so 0 is also a valid address, we had better treat it as normal address when converting it to native address. If you want to treat it as NULL, could you please convert the native pointer into app address and then check whether the app address is 0 instead? e.g.:
```C
foo(wasm_exec_env_t exec_env, void *ptr)
{
    wasm_module_inst_t module_inst = get_module_inst(exec_env);
    uint32 offset = addr_native_to_app(ptr);
    if (offset == 0) // treat as NULL pointer
       report error;
    ...
}
```
This should be OK for wasi-sdk toolchain as the linear memory's offset 0-7 are reserved as in the wasm generated, the first offset of init data is 8.

Paril, 2021-02-22:
Gotcha, the last sentence was what I was interested in (whether 0 was indeed suitable for a "null" offset) since I wasn't sure if it was legal for legitimate memory to live at that address.

I don't recall this being noted in the docs, so it might be worth noting that if you use * that NULL won't expand to zero. I ran into the gotcha real quick in my case and worked around it with the fix you describe!

closed 2021-02-22 by Paril