wasm-micro-runtime/wasm-micro-runtime #539: malloc() in WASI (dlmalloc) throwing "out of bounds memory access"

Paril, 2021-02-22:
I've run into an odd issue with WASI where its implementation of malloc appears to throw an out of bounds memory access. In one instance, I managed to also run into a specific case where the heap grow routine in WAMR was stuck in an infinite loop, as one of the memory relocation sizes was zero, which caused a `while` loop to never exit (again related to malloc, though). This was in ems_kfc.c :: gc_migrate; `size = hmu_get_size(cur);` was getting me zero.

Are there any particular gotchas that should be noted when using malloc? The memory that I give to WAMR from wasm_runtime_native_malloc seems to be in tact, but at some point during runtime malloc() is causing something to break. I'm not using malloc in such a way where the native app is depending on pointers returned from it or anything, so it's not a case where migration is changing addresses on me (the only allocations that I keep track of "permanently" are run through wasm_runtime_native_malloc). In this particular case, it's the second `fopen` that causes the issue to be thrown. In the other case I ran into this bug with, it was the first `malloc` of 16 bytes that caused the crash consistently.

Paril, 2021-02-22:
Small update: all I really have to do is this, in my WASM assembly's init function, to completely break future code:

```
	void *p = malloc(16);
	free(p);
```

The crash then occurs in strlen() while it attempts to count the size of my preopen folder:
```
strlen
__wasilibc_find_abspath
__wasilibc_find_relpath_alloc
__wasilibc_open_nomode
fopen
```

Stepping through my implementation of `fd_prestat_dir_name`, I copy the five bytes of the path name (in my case it's "wasm\0") to the buffer sent to me from WASM. I even called `wasm_runtime_validate_app_str_addr` on the pointer, and verified that all five bytes are within WASM memory.

xujuntwt95329, 2021-02-23:
Hi, this is really strange, and I have never encountered such case. 

According to your description, you are using WASI in your application, so the memory management should be done by wasi-sdk. 
Have you delete or change any files in the wasi-sdk? Could you please download a new wasi-sdk and try again?

Paril, 2021-02-23:
Oh! I have figured it out, actually. I'm sorry, I should have closed this issue.
For those who come across it in the future: the issue was that I compiled my .wasm with the default compiler stack size (which was 216992 bytes), but having an 8 MB stack size set in WAMR. This caused WAMR to allow the WASM code to smash into data, and it just so happened that the dlmalloc global structures and WASI preloads were right at the end of the data section.

This is what tipped me off: https://github.com/WebAssembly/wasi-libc/issues/233

Is this something that WAMR could somehow detect (the stack "overflowing" into the data section)?

wenyongh, 2021-02-24:
@Paril WAMR can detect the auxiliary stack boundary in interpreter mode currently, please add `-Wl,--export=__heap_end -Wl,--export=__data_base` for wasi-sdk clang, so that runtime can resolve the auxiliary stack info. And for malloc/free function, we had better also export them with `-Wl,--export=malloc -Wl,--export=free`, as for wask-sdk-12.0, it might overwrite the app heap managed by WAMR while initializing the heap managed by libc, we might not be able to allocate memory with module_malloc from the app heap. And if the malloc/free functions are exported, the module_malloc/free function will allocate/free memory from/to the heap managed by libc. So we suggest to use wasi-sdk-12.0, and then add flags while compiling wasm app:
```bash
-Wl,--export=__heap_end -Wl,--export=__data_base -Wl,--export=malloc -Wl,--export=free
```

Paril, 2021-02-24:
Cool, I'll make these changes to my project.

closed 2021-02-24 by Paril

wenyongh, 2021-02-24:
@Paril sorry, typo, should be __heap_base and __data_end:
```bash
-Wl,--export=__heap_base -Wl,--export=__data_end -Wl,--export=malloc -Wl,--export=free
```

Paril, 2021-02-25:
Ahh there she goes! Lovely! I'll step through to make sure these are being picked up properly.