wasm-micro-runtime/wasm-micro-runtime #542: Debug memory leak about wasm app

JavanZhu, 2021-02-24:
I noticed the memory consumption of my wasm app is keep growing when running, and in the runtime, memory allocation call stack is following:
```
#01 wasm_runtime_realloc_internal at wasm_memory.c:153
#02 wasm_runtime_realloc at wasm_memory.c:182
#03 aot_enlarge_memory at aot_runtime.c:1506
#04 wasm_runtime_enlarge_memory at wasm_runtime_common.c:1614
```
It's look likes that memory leak is happened in wasm module heap, I would like to know if there are any tips on debugging memory leak like this, such as how to dump memory heap of wasm app.. thank you.

* Language: AssemblyScript

Paril, 2021-02-24:
There's a function that will dump the call stack out of the execution environment (if compiled with the DUMP_STACK feature enabled) - that's usually what I've done to find where allocations are happening.

wenyongh, 2021-02-25:
Hi, the runtime can detect the memory leaks by enabling the heap verification feature, please modify core/config.h, set `BH_ENABLE_GC_VERIFY` to 1. Note that there may be two heaps if the runtime allocator is set to allocate memory from a global heap buffer: the global heap and the app heap. Both of their memory leaks can be detected. And as app heap's space is inside the app's linear memory, it is not an issue in fact if there is a memory leak in app heap, since its space will be freed when the linear memory is destroyed.

JavanZhu, 2021-02-25:
Hi, @wenyongh 
I have set BH_ENABLE_GC_VERIFY to 1, and nothing output, sorry I'm not familar with it, but I will try it more.

I got following assemblyscript call stack of my module depends on @Paril 's reply (Thank you for @Paril's tip), 
```
#00 ~lib/rt/tlsf/growMemory 
#01 ~lib/rt/tlsf/allocateBlock 
#02 ~lib/rt/tlsf/__alloc 
#03 ~lib/rt/pure/__new 
#04 ~lib/array/Array<~lib/string/String>#constructor 
#05 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#readString 
#06 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseKey 
#07 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseObject 
#08 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseValue 
#09 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseArray 
#10 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseValue 
#11 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseObject 
#12 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseValue 
#13 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseArray 
#14 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#parseValue 
#15 ~lib/assemblyscript-json/decoder/JSONDecoder<~lib/assemblyscript-json/JSON/Handler>#deserialize 
#16 ~lib/assemblyscript-json/JSON/_JSON.parse<~lib/string/String> 
#17 samples/memory_leak_as_json_parse/onLoad 
```

I reproduced this problem with a patch [memory_leak_as_json_parse.patch.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6043029/memory_leak_as_json_parse.patch.zip) (a AssemblyScript demo project and some code), could you please have a try? thank you.

## step
* build wasm
npm run build:memory_leak_as_json_parse
* run wasm module
./iwasm -f onLoad ../memory_leak_as_json_parse.wasm
* input any character to call onLoad() method which will call JSON.parse
* and the consumption of wasm module will print like following, the total memory size is keep growing, not sure it's memory leak or other problem.
```
WASM module inst memory consumption, total size: 138616
WASM module inst memory consumption, total size: 269688
WASM module inst memory consumption, total size: 531832
WASM module inst memory consumption, total size: 531832
...
WASM module inst memory consumption, total size: 2104696
```

JavanZhu, 2021-02-25:
> Hi, the runtime can detect the memory leaks by enabling the heap verification feature, please modify core/config.h, set `BH_ENABLE_GC_VERIFY` to 1. Note that there may be two heaps if the runtime [...]

I'm not familiar with the runtime memory layout, could you please tell me some information like diagram about it. And I understand wether "linear memory is destroyed" means the entire wasm module is destroyed? If so, my wasm app will continue to run after loading it, so it will never be free.

wenyongh, 2021-02-26:
Hi @JavanZhu, the memory leak info will be dumped if it is detected when the heap is destroyed: (1) app heap is destroyed when wasm_runtime_deinstantiate is called, but in fact the app heap isn't enabled as module_malloc allocates memory from the heap managed by the assembly script app, by calling the __new function exported by app, (2) global heap is destroyed when wasm_runtime_destroy is called, and you should set the runtime memory allocator with global heap mode, you can set it by changing macro USE_GLOBAL_HEAP_BUF to 1 in main.c. For your test case, after you press 'q' to exit loop, we can test that the exec_env created by yourself isn't destroyed. And normally you can also use `valgrind` tool to detect the memory leak: set USE_GLOBAL_HEAP_BUF to 0, build iwasm with debug mode (cmake .. -DCMAKE_BUILD_TYPE=Debug), and then run
```bash
valgrind --leak-check=full ./iwasm -f onLoad memory_leak_as_json_parse.wasm
```
After fixing the exec_env not destroyed issue introduced by yourself, there is no other memory leaks detected. The memory consumption keeps growing since the wasm app keeps allocating memory, when the linear memory size isn't enough, runtime will reallocate another bigger memory. And when runtime isn't able to reallocate a bigger memory, the wasm app throws `unreachable` exception, you can set USE_GLOBAL_HEAP_BUF to 1 with smaller size e.g. 1M to reproduce it. Thanks.

JavanZhu, 2021-02-26:
Hi @wenyongh.
yes, my demo patch has a memory leak because I forgot to call wasm_runtime_destroy_exec_env() to destroy exec_env, sorry . 

In my AssemblyScript case, the reason of memory consumption grow is the following snippet code:
```AssemblyScript
    let weather: JSON.Arr = <JSON.Arr>JSON.parse(mockJsonData);
```
So I understand this problem may be caused by [AssemblyScript-json lib](https://github.com/nearprotocol/assemblyscript-json), WAMR runtime can not know wether there is some memory leak because the heap is managed by AssemblyScript app, right? Thanks.

wenyongh, 2021-03-01:
@JavanZhu no, runtime can not know memory leak info of the heap managed by AssemblyScript library.

closed 2021-03-02 by JavanZhu