wasm-micro-runtime/wasm-micro-runtime #547: About  lib pthread supported on windows

bxkai, 2021-02-27:
In [win_thread.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/windows/win_thread.c)，the interface implementations are empty. How to support lib pthread on windows？

wustwn, 2021-03-01:
Hi, currently we haven't supported lib pthread on windows.

bxkai, 2021-03-13:
@wustwn Hi，If we want to support lib pthread on windows, pthread-win32 may be a good choice, any suggestions? Thank you！

wustwn, 2021-03-16:
Hey, 

You can give it a try, however, pthread-win32's license is `LGPL` while WAMR's license is `Apache 2.0 license` with the LLVM exception, which means the patch can not be accepted by wamr. 

Thanks.

bxkai, 2021-04-02:
@wustwn windows uvwasi depends [libuv](https://github.com/libuv/libuv), libuv’s thread API is analogous to the pthreads API.  lib pthread can use libuv’s thread?

wenyongh, 2021-06-09:
@bxkai the lib-pthread for Windows is supported now, you can have a try, note that you should add -DWAMR_BUILD_LIB_PTHREAD=1 when building iwasm, and add --enable-multi-thread for wamrc when generating AOT file. Thanks.

bxkai, 2021-06-15:
@wenyongh  I have a try and the exception appeared.  [The exception code position](https://github.com/bytecodealliance/wasm-micro-runtime/blob/1a4aa5ac2f659439a344c0ec7163b157c0e190ce/core/shared/platform/windows/win_thread.c#L204), the "parent" is null poniter.   I think the reason is call "wasm_runtime_call_wasm" function in some different host thread. 
    1) the first host thread A init env:
wasm_runtime_env_init->bh_platform_init->os_thread_sys_init->TlsSetValue(thread_data_key)
   2) the second host thread  B call wasm to create thread.  thread_data_current->TlsGetValue(thread_data_key) , thread_data_key should be null

wenyongh, 2021-06-16:
The thread to call wasm function to create a new thread should be managed by runtime but not the standalone host thread which is created by host embedder (e.g. _beginthreadex on Windows or pthread_create on Linux) but not runtime. Could you please try using [wasm_runtime_spawn_thread()](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/include/wasm_export.h#L861-L863) instead of _beginthreadex like function to create the new thread and call wasm function in the new thread? Ref to the [spawn-thread sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/spawn-thread) for more details.

bxkai, 2021-06-16:
@wenyongh wasm_runtime_spawn_thread()  is binded with a wasm instance，each wasm instance must using  wasm_runtime_spawn_thread()  to create new thread alonely. So some different wasm instances can not shared same thread, and if having many wasm instances, the number of thread is considerable

wenyongh, 2021-06-16:
So you want to run multiple wasm instances in different threads, and in their own thread, call wasm function to create instance's own thread? Could you please try firstly calling os_thread_create() in the main thread initialized by wasm_runtime_init/full_init()  to create the new thread to run another wasm instance, and then in the new thread, calling wasm function to create the instance's own thread?

bxkai, 2021-06-24:
@wenyongh I try change [thread_data_current](https://github.com/bytecodealliance/wasm-micro-runtime/blob/1a4aa5ac2f659439a344c0ec7163b157c0e190ce/core/shared/platform/windows/win_thread.c#L111) as below，It works well calling "wasm_runtime_call_wasm" function in some different host thread. Is there any risk in this modification? Thanks！

thread_data_current()
{
    os_thread_data *current = (os_thread_data *)TlsGetValue(thread_data_key);
    if (current == NULL) {
        current = &supervisor_thread_data;
    }
    return current;
}

wenyongh, 2021-06-25:
Hi, the supervisor_thread_data is the thread data of the runtime main thread, normally it should not be used by other thread. For example, if the two threads A and B with same thread data are calling os_cond_wait/timedwait to wait other threads simultaneously, then the same thread data's semaphore is used to wait, suppose A waits for thread C, B waits for thread D, then if C notifies A, B might be notified also (or B is notified but A isn't), so the behavior is unexpected.

We had better use os_thread_create to create a new thread to run a new instance, if you prefer not to do that, then normally you should attach/bind you host thread to the runtime to create the necessary data structure, like Java JNI's [AttachCurrentThread API](https://docs.oracle.com/javase/9/docs/specs/jni/invocation.html#attachcurrentthread). We roughly added two APIs os_thread_attach_current_thread and os_thread_detach_current_thread in windows platform, see attachment. And before running wasm app, you should call os_thread_attach_current_thread(), and after wasm app ends, you should call os_thread_detach_current_thread(). The patch is just for a reference, could you please test it by yourself? Thanks.

[win_thread.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6714897/win_thread.zip)

bxkai, 2021-06-28:
@wenyongh  We need to support native app and wasm app. We have a module framework only supports native apps，and build a module wasm-app-runtime based on wamr, which is dynamically loaded by the framework. The module framework manges thread's creating and destory,  the module wasm-app-runtime do not have thread management. As your suggestion，wasm-app-runtime  need to create thread for every wasm app,  and I try to call os_thread_attach_current_thread and os_thread_detach_current_thread to bing wasm app thread, it's work well. But for us,module wasm-app-runtime increased thread management and increased complexity .

wenyongh, 2021-08-03:
@bxkai there is really no good idea that runtime can auto init/destroy thread environment if developer creates the thread by himself but not runtime. To unify the runtime APIs, we implemented os_thread_env_init/destroy() for Windows platform instead of attach/detach_current_thread(), and they are called in wasm_runtime_init_thread_env()/wasm_runtime_destroy_thread_env(), which are exported APIs in wasm_export.h, see #683. Could you please pull latest code and call wasm_runtime_init_thread_env()/wasm_runtime_destroy_thread_env() instead? Thanks.

closed 2021-11-01 by wenyongh