wasm-micro-runtime/wasm-micro-runtime #549: [Bug] x86_32 AOT crashes on Zephyr Posix build

AfflatusX, 2021-02-27:
Using the latest release, the AOT mode + AssemblyScript AOT binary crashes when installing the app. After poking around, the crash seems to happen at: https://github.com/bytecodealliance/wasm-micro-runtime/blob/79afa493aa9895cffeeefbc0b9020bb2020a3ae8/core/iwasm/aot/aot_runtime.c#L713

Interpreter mode works fine. 

You can use the repro branch mentioned in https://github.com/bytecodealliance/wasm-micro-runtime/issues/441 to validate that. I applied the fix proposed in that issue. The WAMR version in that is still Sep 2020, but I tried both they crash in the same way.

Let me know if there is any way I can help to repro/debug this issue!

lum1n0us, 2021-03-02:
please try with *qemu_x86_nommu* instead of *native_posix*.

AfflatusX, 2021-03-12:
Unfortunately some API we rely on does not seem to be supported by qemu_x86_nommu. 

I guess that's a known issue then?

lum1n0us, 2021-03-17:
Different board/mode might have totally different result. For example, we can get executable memory in  *qemu_x86_no_mmu* but not in *qemu_x86*. *stm32* is available but others are not. And I didn't find any document have records on it so far.

closed 2021-06-09 by wenyongh