wasm-micro-runtime/wasm-micro-runtime #551: How to correctly passing byte array from native to WASM app.

AfflatusX, 2021-03-01:
I'm using 09-29-2020 version of WAMR, my WASM app is written in AssemblyScript. WAMR running in interpreter mode.

Context:

I have a blob data (around 16 KB but could be bigger than that) which I want to pass it to AS side to process. I used wasm_runtime_module_dup_data to copy the data into WASM memory space and obtain the offset. Then I return the offset and length by encoding them into an i64 type.

Problem:

On the AS side, I'm able to decode the offset and length properly. I use changetype<ArrayBuffer>(ptr) to convert it into an ArrayBuffer type. However, querying the length of the ArrayBuffer yields garbage data. Note that this behavior does not seem to be deterministic - I was able to cast it with correct size at some point, but it's not reliable.

I noticed in the sample AS code, we manually copy those bytes to a new array with specified size. I'm wondering if there is anyway to avoid that due to the memory cost (even if it's just temporary). My understanding is that, by using wasm_runtime_module_dup_data, it already allocates the memory in app space for the WASM app to access.

Any suggestion would be great, thanks in advance!

xujuntwt95329, 2021-03-01:
AssemblyScript has its own runtime, memory management mechanism, and GC. The ArrayBuffer is an object managed by AS runtime, you can find its layout [here](https://www.assemblyscript.org/memory.html#internals). The `wasm_runtime_module_dup_data`, however, just returns the raw pointer of the buffer, I‘m not sure whether AssemblyScript could construct an object from a raw pointer.

Maybe you need to export some API from AS which could create the ArrayBuffer and write the data into it from the native side. The AssemblyScript's [loader](https://www.assemblyscript.org/loader.html#installation) demonstrated the interaction between AS and JavaScript, but you may also use such mechanism between AS and C.

AfflatusX, 2021-03-01:
Ah, thanks for the reply - this makes sense! I'll take another look - this is very helpful!

closed 2021-03-01 by AfflatusX