wasm-micro-runtime/wasm-micro-runtime #552: Question about aee_host_msg_callback payload size

AfflatusX, 2021-03-01:
Right now aee_host_msg_callback's msg_len is a 16bit unsigned integer:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/282831eba5ac1b19c6d535a4c60a15d935e13e60/core/app-mgr/app-manager/app_manager_host.c#L194

The call fails when the payload size overflows without any error message - essentially install app request hangs.

Is there any specific reason that this can't be a 32 bit integer?

wustwn, 2021-03-04:
Hi, thanks for reporting this issue! It has been fixed in patch [#555](https://github.com/bytecodealliance/wasm-micro-runtime/pull/555), could you please try again?

AfflatusX, 2021-03-04:
Thanks for the fix! I'll give it a try tomorrow.

closed 2021-03-12 by AfflatusX