wasm-micro-runtime/wasm-micro-runtime #554: Question about Memory Profiling

AfflatusX, 2021-03-02:
I turned on Memory Profiling and Tracing and here is the break-down of memory usage from WAMR/App:

WASM module memory consumption, total size: 161409
    module struct size: 188
    types size: 379
    imports size: 1632
    funcs size: 2216
    tables size: 16
    memories size: 16
    globals size: 1320
    exports size: 132
    table segs size: 136
    data segs size: 7596
    const strings size: 8642
    aot code size: 139136
Exec env memory consumption, total size: 12344
    exec env struct size: 56
    stack size: 12288
WASM module inst memory consumption, total size: 69296
    module inst struct size: 416
    memories size: 65844
        app heap size: 0
    tables size: 100
    functions size: 2624
    globals size: 132
    exports size: 180

I'm wondering if there is some strategy I can adapt to reduce AOT Code Size? (The code is written in AssemblyScript, so it's runtime might be playing a big role here).

Also, WASM module inst memory consumption seems to be fixed - is that something we can adjust, or it's related to WASM page size?

Any suggestion on memory optimization would be great!

wustwn, 2021-03-04:
Hi, there are some configurations which might be useful, you can ref to them by running `asc --help`: 

```bash

Optimization
  --optimize, -O        Optimizes the module. Typical shorthands are:

                         Default optimizations   -O / -O3s
                         Make a release build    -O --noAssert
                         Make a debug build      --debug
                         Optimize for speed      -O3
                         Optimize for size       -O3z --converge

  --optimizeLevel       How much to focus on optimizing code. [0-3]
  --shrinkLevel         How much to focus on shrinking code size. [0-2, s=1, z=2]
  --converge            Re-optimizes until no further improvements can be made.
  --noAssert            Replaces assertions with just their value without trapping.
  --runtime             Specifies the runtime variant to include in the program.

                         full  Default runtime based on TLSF and reference counting.
                         half  Same as 'full', but not exported to the host.
                         stub  Minimal stub implementation without free/GC support.
                         none  Same as 'stub', but not exported to the host.

--initialMemory       Sets the initial memory size in pages.
--maximumMemory       Sets the maximum memory size in pages.

```

You can consider setting -O3z optimization size, and set --runtime half or --runtime stub, which will not include all assembly script runtime in the wasm application. Besides, you can use --initialMemroy and --maximumMemory respectively to set the initial and maximum page count of the wasm application.

AfflatusX, 2021-03-04:
@wustwn - I see. Stub runtime indeed reduces the size but without free it's not going to scale - how is heap memory free handled in that case? (I can ask in AS community, but just in case it is relevant to how WAMR works)

is the WASM module inst memory referring to the page size of WebAssembly (64 KB)?

xujuntwt95329, 2021-03-04:
> how is heap memory free handled in that case?

Hi, as AssemblyScript has its own runtime, the memory is managed by itself, and not visible to WAMR. So if the sub runtime don't free the memory, WAMR won't do this, either. Seems AssemblyScript support [Custom runtime](https://www.assemblyscript.org/garbage-collection.html#custom-runtime) which allows you to use your own memory manager, but this may introduce a lot of work to do.

> is the WASM module inst memory referring to the page size of WebAssembly (64 KB)?

"WASM module inst memory" means the memory consumption of the internal data structure `WASMModuleInstance`, the linear memory is just part of it, there are also functions, tables, globals in that structure.

AfflatusX, 2021-03-12:
Closing. Thanks for the answer!

closed 2021-03-12 by AfflatusX