wasm-micro-runtime/wasm-micro-runtime #556: Question about "out of bounds memory access"

JavanZhu, 2021-03-04:
Hi, sometimes runtime thrown "out of bounds memory access" Exception when running my AssemblyScript wasm app, could you please give me some tips about how to debug such kind of error? I am also trying to reproduce this bug. Thanks

xujuntwt95329, 2021-03-04:
You can build WAMR with `WAMR_BUILD_DUMP_CALL_STACK` enabled, then the call stack will be printed out after the exception occurs. Please refer to [dump call stack](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#enable-dump-call-stack-feature) for more details.

JavanZhu, 2021-03-09:
Hi, @xujuntwt95329, thank you for your help, I got following AssemblyScript call stack:
```
#00 ~lib/rt/tlsf/allocateBlock 
#01 ~lib/rt/pure/__new 
#02 ~lib/string/String.__concat 
#03 ~lib/util/string/joinReferenceArray<~lib/string/String> 
#04 ~lib/array/Array<~lib/string/String>#join 
```

I figured out this exception came from
https://github.com/bytecodealliance/wasm-micro-runtime/blob/a1568825e85f010cec27e0cf6e88617846481d9b/core/iwasm/interpreter/wasm_interp_classic.c#L1601-L1614
https://github.com/bytecodealliance/wasm-micro-runtime/blob/a1568825e85f010cec27e0cf6e88617846481d9b/core/iwasm/interpreter/wasm_interp_classic.c#L229-L237

and I print some variable value for debugging purposes. full log and log position in source code can be found [ here out-of-bounds-memory-access.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6107568/out-of-bounds-memory-access.zip).

----
```
// ignore some log
2021-03-09 17:05:46.424 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [65536], linear_mem_size addr = [0xa8ed962c]
2021-03-09 17:05:46.436 31248-31292/ I/wasm_runtime::: first return value argv[0]=41296
2021-03-09 17:05:46.437 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:46.437 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [65536], linear_mem_size addr = [0xa8ed9644]
2021-03-09 17:05:46.708 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:46.709 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [65536], linear_mem_size addr = [0xa8ed833c]
2021-03-09 17:05:46.712 31248-31292/ I/wasm_runtime::: first return value argv[0]=48544
// some duplicate log...
2021-03-09 17:05:47.734 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:47.734 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [65536], linear_mem_size addr = [0xa8ed7684]
2021-03-09 17:05:47.742 31248-31292/ I/wasm_runtime::: WASM_OP_MEMORY_GROW: linear_mem_size = [131072], linear_mem_size addr = [0xa8ed7684]
2021-03-09 17:05:47.796 31248-31292/ I/wasm_runtime::: no return value
2021-03-09 17:05:47.796 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:47.796 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [131072], linear_mem_size addr = [0xa8ed7644]
2021-03-09 17:05:47.796 31248-31292/ I/wasm_runtime::: no return value
2021-03-09 17:05:47.797 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:47.797 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [131072], linear_mem_size addr = [0xa8ed7644]
2021-03-09 17:05:47.797 31248-31292/ I/wasm_runtime::: no return value
2021-03-09 17:05:47.797 31248-31292/ I/wasm_runtime::: it is a function of the module itself
// some duplication log
2021-03-09 17:05:47.978 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [131072], linear_mem_size addr = [0xa8ed7e94]
2021-03-09 17:05:47.978 31248-31292/ I/wasm_runtime::: no return value
2021-03-09 17:05:47.979 31248-31292/ I/wasm_runtime::: it is a function of the module itself
2021-03-09 17:05:47.979 31248-31292/ I/wasm_runtime::: wasm_interp_call_func_bytecode: linear_mem_size = [131072], linear_mem_size addr = [0xa8ed7e94]
2021-03-09 17:05:47.979 31248-31292/ I/wasm_runtime::: no return value
**2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: CHECK_MEMORY_OVERFLOW: op = [WASM_OP_F32_LOAD]
2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: CHECK_MEMORY_OVERFLOW: offset = [0], addr = [66540]
2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: CHECK_MEMORY_OVERFLOW: linear_mem_size = [65536], linear_mem_size addr = [0xa8ed9644]
2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: CHECK_MEMORY_OVERFLOW: linear_mem_size addr = [0xa8ed9644]**
2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: **** wasm code crash dump ****
2021-03-09 17:05:48.084 31248-31292/ I/wasm_runtime::: #00 ~lib/rt/tlsf/allocateBlock 
2021-03-09 17:05:48.085 31248-31292/ I/wasm_runtime::: #01 ~lib/rt/pure/__new 
2021-03-09 17:05:48.085 31248-31292/ I/wasm_runtime::: #02 ~lib/string/String.__concat 
2021-03-09 17:05:48.085 31248-31292/ I/wasm_runtime::: #03 ~lib/util/string/joinReferenceArray<~lib/string/String> 
2021-03-09 17:05:48.085 31248-31292/ I/wasm_runtime::: #04 ~lib/array/Array<~lib/string/String>#join 
2021-03-09 17:05:48.086 31248-31292/ I/wasm_runtime::: **** **** **** ***** **** ****
2021-03-09 17:05:48.086 31248-31292/ I/wasm_runtime::: meet an exception Exception: out of bounds memory access
```
In function CHECK_MEMORY_OVERFLOW(),  I noticed that the value of variable linear_mem_size is still equal previous value 65536 after WASM_OP_MEMORY_GROW. Could this be the cause of this exception? 
Please tell me if you need to add more logs, thanks.

wenyongh, 2021-03-09:
@JavanZhu Thanks for finding the issue! Seems that the memory.grow opcode was executed in the native API, which called wasm_runtime_module_malloc function to allocate memory from assemblyscript's heap, and after returning from the native API, interpreter updates memory ptr without memory size. I uploaded a patch to fix it, could you please again for both fast-interp and classic-interp?
[update_memory.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6108184/update_memory.zip)

JavanZhu, 2021-03-10:
Hi @xujuntwt95329, I have tested my wasm app in classic-interp and fast-interp, both works well, this issue has been fixed. Thanks.

closed 2021-03-10 by JavanZhu