wasm-micro-runtime/wasm-micro-runtime #558: Crash in wasm_instance_exports using wasm_c_api with memory exports

ScottTodd, 2021-03-05:
I'm trying to run a wasm module produced by my compiler project through WAMR's WASM C API, based on the samples here: https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/wasm-c-api. For context, see https://github.com/google/iree/issues/2863.

[>> This gist <<](https://gist.github.com/ScottTodd/350ea0d404493f876beecad0a1e51d7e) has my
* `.wat` file
* the `.c` code I'm trying to run it with (side-by-side [here](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/wasm-c-api/src) with the other WAMR WASM C API examples)
* the source `.mlir` that my compiler project used to generate the `.wat` file

When I run the program, it crashes on `wasm_instance_exports(instance, &exports)`.

Here is the relevant part of the call stack, where `orig` is `NULL`: https://github.com/bytecodealliance/wasm-micro-runtime/blob/54e82ec439552018d0f26c60b0fd6cc9e36b42a8/core/iwasm/common/wasm_c_api.c#L2752-L2754

that happens after https://github.com/bytecodealliance/wasm-micro-runtime/blob/54e82ec439552018d0f26c60b0fd6cc9e36b42a8/core/iwasm/common/wasm_c_api.c#L2237-L2246

appends `external` when it is still `NULL` to the exports.

Full callstack:
```
iree_add.exe!wasm_extern_kind(const wasm_extern_t * extrenal) Line 2692 (d:\dev\projects\iree\third_party\wasm-micro-runtime\core\iwasm\common\wasm_c_api.c:2692)
iree_add.exe!wasm_extern_copy(const wasm_extern_t * src) Line 2641 (d:\dev\projects\iree\third_party\wasm-micro-runtime\core\iwasm\common\wasm_c_api.c:2641)
iree_add.exe!wasm_extern_vec_copy(wasm_extern_vec_t * out, const wasm_extern_vec_t * src) Line 2756 (d:\dev\projects\iree\third_party\wasm-micro-runtime\core\iwasm\common\wasm_c_api.c:2756)
iree_add.exe!wasm_instance_exports(const wasm_instance_t * instance, wasm_extern_vec_t * out) Line 2613 (d:\dev\projects\iree\third_party\wasm-micro-runtime\core\iwasm\common\wasm_c_api.c:2613)
iree_add.exe!main(int argc, const char * * argv) Line 95 (d:\dev\projects\iree\third_party\wasm-micro-runtime\samples\wasm-c-api\src\iree_add.c:95)
iree_add.exe!invoke_main() Line 79 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:79)
iree_add.exe!__scrt_common_main_seh() Line 288 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288)
iree_add.exe!__scrt_common_main() Line 331 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:331)
iree_add.exe!mainCRTStartup() Line 17 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp:17)
kernel32.dll!00007ffa55667c24() (Unknown Source:0)
ntdll.dll!00007ffa5606d721() (Unknown Source:0)
```

(Building on Windows using CMake with `-DWAMR_BUILD_PLATFORM=windows`, the other samples all work for me)

---

So, I'm not sure if
* I'm doing something wrong in how my project is producing that `.wat` file
* I'm doing something wrong in how I'm using WAMR's WASM C API implementation
* WAMR's WASM C API implementation is missing support for something (exported memory?)

I'm pretty new to the WASM tools / ecosystem, so can someone help me out here?

lum1n0us, 2021-03-05:
Yes you are right. WAMR doesn't support export MEMORY and TABLE temporarily.

benvanik, 2021-03-05:
Thanks for the response! Does that mean there's no way to use the equivalent of `wasm_runtime_module_malloc` via the wasp-c-api (or emulate it) today? We need to marshal in strings and buffers via the wasm heap and are trying to effectively do what this is doing: https://marcoselvatici.github.io/WASM_tutorial/#memory

ScottTodd, 2021-03-05:
Thanks!

I'm shifting to look at https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/basic instead of the WASM C API now, and I noticed that even that example exports memory (built on my machine): https://gist.github.com/ScottTodd/47237ebc73a0633cdfcfaaf474de5780#file-testapp-wasm-L566-L569

To build, I modified https://github.com/bytecodealliance/wasm-micro-runtime/blob/54e82ec439552018d0f26c60b0fd6cc9e36b42a8/samples/basic/build.sh#L44-L54 

to just use `clang` , which on my machine is installed as:
```
λ clang --version
clang version 11.0.0
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin
```

(maybe the [WASI-SDK](https://github.com/WebAssembly/wasi-sdk) clang sets different default compile options, but I doubt it?)

---

I also tried the `reverse` example from https://ariya.io/2019/05/basics-of-memory-access-in-webassembly using `emcc` version `2.0.9 (59aedd441c317b7064146d4d888eab272e92dd03)` and see exported memory there too.

ScottTodd, 2021-03-06:
Nice, the basic sample (using the iwasm VM core instead of the WASM C API) is working with my original wasm module. Messy rough draft: https://gist.github.com/ScottTodd/4d01f92c434e6caebf1bf2b180f02840

closed 2021-06-29 by wenyongh

Soliloquy0810, 2022-03-08:
May I ask a simple question? I wonder how you output the full callstack in this issue?

ScottTodd, 2022-03-08:
> May I ask a simple question? I wonder how you output the full callstack in this issue?

This was a while ago, so I probably either copied from the console with a debug build (`-DCMAKE_BUILD_TYPE=Debug`) or I copied the callstack while running under a debugger (e.g. https://code.visualstudio.com/docs/editor/debugging).