wasm-micro-runtime/wasm-micro-runtime #561: Embedding wamr in Rust with wasi

PhilippGackstatter, 2021-03-07:
Hello,

I'm embedding wamr in a Rust application, and I'm using [wamr_sys](https://crates.io/crates/wamr-sys) for the bindings. I'm unable to get wasi working, specifically simply printing to stdout or passing arguments to the Wasm module.

My WebAssembly module is this Rust code:

```rust
pub fn main() {
    let args: Vec<String> = std::env::args().collect();

    println!("number of args {:?}", args.len());
}
```
which I'm compiling with:

```
cargo build --release --target wasm32-wasi --example test_module
```

and I've stripped down my Rust code to a minimal working example that has the mentioned issue.

```rust
use std::{
    ffi::{CStr, CString},
    os::raw::c_char,
};

use wamr_sys::*;

fn main() -> anyhow::Result<()> {
    let wasm_module_bytes = include_bytes!(
        "/path/to/test_module.wasm"
    );
    let mut error_buf: Vec<i8> = vec![32; 128];
    const STACK_SIZE: u32 = 8092;

    unsafe {
        if !wasm_runtime_init() {
            anyhow::bail!("runtime_init returned false");
        }

        let module = wasm_runtime_load(
            wasm_module_bytes.as_ptr(),
            wasm_module_bytes.len() as u32,
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        );

        if module.is_null() {
            panic!(
                "Module is null: {}",
                CStr::from_ptr(error_buf.as_ptr()).to_string_lossy()
            );
        }

        let args = vec![CString::new("test").unwrap()];

        let c_args = args
            .iter()
            .map(|arg| arg.as_ptr())
            .collect::<Vec<*const c_char>>();

        let null_ptr = 0 as *mut *const i8;

        wasm_runtime_set_wasi_args(
            module,
            null_ptr,
            0,
            null_ptr,
            0,
            null_ptr,
            0,
            c_args.as_ptr() as *mut *mut i8,
            c_args.len() as i32,
        );

        let module_inst = wasm_runtime_instantiate(
            module,
            STACK_SIZE,
            0,
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        );

        if module_inst.is_null() {
            panic!(
                "Instatiated module is null: {}",
                CStr::from_ptr(error_buf.as_ptr()).to_string_lossy()
            );
        }

        let exec_env = wasm_runtime_create_exec_env(module_inst, STACK_SIZE);

        if exec_env.is_null() {
            panic!(
                "Exec env is null: {}",
                CStr::from_ptr(error_buf.as_ptr()).to_string_lossy()
            );
        }

        let start_func = wasm_runtime_lookup_wasi_start_function(module_inst);

        if start_func.is_null() {
            panic!("start func is null");
        }

        let mut results: Vec<wasm_val_t> = vec![];
        let mut args: Vec<wasm_val_t> = vec![];

        if wasm_runtime_call_wasm_a(
            exec_env,
            start_func,
            results.len() as u32,
            results.as_mut_ptr(),
            args.len() as u32,
            args.as_mut_ptr(),
        ) {
            println!("success");
        }
    }

    Ok(())
}
```

If I run this, it prints success but nothing else.

However, if I run this with `iwasm`, I get this:

```shell
$ ./iwasm /path/to/test_module.wasm hello
number of args 2
```

as expected. I must be doing something wrong in the Rust setup, but I can't figure out what it is. The library is compiled with [this CMakeLists.txt](https://github.com/rahul-thakoor/wamr-sys/blob/main/libiwasm/CMakeLists.txt), which is identical in terms of the enabled features to the one in `product-mini/platforms/linux/CMakeLists.txt`.

Is there anything obvious I'm doing wrong?

Thanks a lot for your help!

xujuntwt95329, 2021-03-09:
Hi, @PhilippGackstatter, thanks for providing the minimal example. 
I noticed that you instantiate the wasm module without app heap:

``` Rust
        let module_inst = wasm_runtime_instantiate(
            module,
            STACK_SIZE,
            0,                        // <---- heap_size
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        );
```

This will avoid WAMR to create the `wasi-ctx` for this instance, so the app will fail while calling WASI APIs:

``` bash
#00 proc_exit 
#01 _Exit 
#02 __wasilibc_initialize_environ          <---- call WASI API in this function, and got fail result since wasi-ctx not exists
#03 __wasilibc_initialize_environ_eagerly 
#04 __wasm_call_ctors 
#05 _start 
```

Add the heap_size during instantiate can solve this problem.
``` Rust
        const HEAP_SIZE: u32 = 1024;
        let module_inst = wasm_runtime_instantiate(
            module,
            STACK_SIZE,
            HEAP_SIZE,
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        );

```

BTW, in WAMR's latest code, the `wasi-ctx` is not allocated on the app heap, I'll check if we can remove this check in the future.

PhilippGackstatter, 2021-03-09:
Thank you for your detailed answer, very helpful!

I was able to get it working by setting the `HEAP_SIZE` to `1024` as you suggested.

The documentation for `wasm_runtime_instantiate` states:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/a1568825e85f010cec27e0cf6e88617846481d9b/core/iwasm/include/wasm_export.h#L334-L337

That's why I set the size to 0 thinking it would create a heap with some default size. Did I misunderstand the documentation or is this perhaps outdated?

I have another question about thread safety. The embed documentation already states that `exec_env` is not thread safe. What about `wasm_module_t`, is that thread safe? Ideally, I would like to load modules once, and then execute them on different threads with different wasi arguments.

Though it seems to me that sharing `wasm_module_t` across threads is not safe, if I use `wasm_runtime_set_wasi_args` to set the arguments, since that modifies the module, here:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/a1568825e85f010cec27e0cf6e88617846481d9b/core/iwasm/common/wasm_runtime_common.c#L1632-L1658

Is that right? Do you have a suggestion for how I could share modules across threads?

xujuntwt95329, 2021-03-16:
> That's why I set the size to 0 thinking it would create a heap with some default size. Did I misunderstand the documentation or is this perhaps outdated?

The documentation is outdated, I've fixed this recently, thanks for pointing it out.

> Ideally, I would like to load modules once, and then execute them on different threads with different wasi arguments.

There may be a workaround for your requirement:
After you create the second instance from the same module, you can use `wasm_runtime_destroy_wasi` to destroy the `wasi-ctx` with old args and use `wasm_runtime_init_wasi` to re-create the `wasi-ctx` with new arguments for your new instance.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/fda3a26903512c13078f47e7c9e082f1b2c7c794/core/iwasm/common/wasm_runtime_common.h#L410-L419

PhilippGackstatter, 2021-04-09:
> After you create the second instance from the same module, you can use `wasm_runtime_destroy_wasi` to destroy the `wasi-ctx` with old args and use `wasm_runtime_init_wasi` to re-create the `wasi-ctx` with new arguments for your new instance.

That doesn't work for me unfortunately, since these functions are not part of `wasm_export.h` and thus are not part of the Rust bindings in `wamr_sys`. I am now copying the module bytes, if they need to be shared with multiple threads, which actually seems to be fast enough.

closed 2021-04-09 by PhilippGackstatter