wasm-micro-runtime/wasm-micro-runtime #567: How can I return a c_str from wasm app to native

xingkaiyu, 2021-03-11:
I am using wamr in android and I want to write a function to pass a string result to native.
So I use char* as the return type like this:
const char *TestReturnString();

Then I have a question:
If I return a local cstring like below, the result is not always right. Is it because the result will be free when the wasm funtion end?
const char *TestReturnString()
{
    std::string result = "12345";
    return result.c_str();
}

And if I return a cstring by malloc and memcpy, I will get the right result. 
But need I free the cstring in native because it is newed. If I need how can I do it? Just use normal free in native to the char* returned or I need to transfer the returned address to wasm by wasm_runtime_addr_native_to_app and use wasm_runtime_module_free function to free it?

wenyongh, 2021-03-14:
Hi, yes, the local variable is temporary variable on stack, please do not use it again after the wasm function ends. And yes, you can allocate a memory and pass it to native, and normally you should free it when it isn't used, or the app heap might be used up. To free the memory in native, if the wasm app is built with non -nostdlib mode (or wasi-libc mode), then firstly please export the malloc/free function when compiling wasm app, e.g. `-Wl,--export=malloc -Wl,--export=free` for wasi-sdk or emsdk. If the wasm app is built with -nostdlib mode, then no need to export the malloc/free function. And then please call wasm_runtime_module_free function in native lib to free it:
```C
wasm_module_inst_t module_inst = get_module_inst(exec_env);
uint32 app_offset = wasm_runtime_addr_native_to_app(module_inst, native_ptr);
wasm_runtime_module_free(app_offset);
```

xingkaiyu, 2021-03-15:
@wenyongh 
Thanks for your answer.
And I want to know that:
1- After I use the wasm_runtime_module_free to free in native, need I use free funtion in native to free the native address again?
2- If I always export malloc and free function in wasm app, is it ok to use wasm_runtime_module_free when I use -nostdlib mode or other sdk from wasi-sdk?

wenyongh, 2021-03-16:
> And I want to know that:
> 1- After I use the wasm_runtime_module_free to free in native, need I use free funtion in native to free the native address again?

No, no need to call free function again.

> 2- If I always export malloc and free function in wasm app, is it ok to use wasm_runtime_module_free when I use -nostdlib mode or other sdk from wasi-sdk?

Yes, you can also export malloc/free function in -nostdlib mode.

xingkaiyu, 2021-03-16:
@wenyongh 
Thanks a lot!

closed 2021-03-16 by xingkaiyu