wasm-micro-runtime/wasm-micro-runtime #571: Is there any protect for shared_ptr or nullptr in wamr

xingkaiyu, 2021-03-13:
Hi~
I am using shared_ptr in wasm, and my sdk is wasi-sdk-12.0.
I wrote a class to hold the buffer and its length:
class CppBytes {
public:
    CppBytes(const char *buffer, uint32_t size) {
        if (nullptr == buffer || 0 == size) {
            mBuffer = nullptr;
            mSize = 0;
        } else {
            mBuffer = (char *)malloc(size);
            memcpy(this->mBuffer, buffer, size);
            this->mSize = size;
        }
    }
    ~CppBytes() {
        if (mBuffer != nullptr) {
            free(mBuffer);
            mBuffer = nullptr;
        }
    }

    const char *GetBuffer() {
        return mBuffer;
    }
    uint32_t GetSize() {
        return mSize;
    }

private:
    char *mBuffer;
    uint32_t mSize;
};

And I wrote a class which has a shared_ptr<CppBytes> member:
class ClassHasCppBytes {
public:
    std::shared_ptr<CppBytes> GetData() {
        return mData;
    }
    void SetData(std::shared_ptr<CppBytes> data) {
        mData = data;
    }

private:
    std::shared_ptr<CppBytes> mData;
};

Then I had the test function like this:
void TestSharedPtrWasm()
{
    std::string s = "123";
    std::shared_ptr<CppBytes> bytes = std::make_shared<CppBytes>(s.c_str(), s.size() + 1);

    ClassHasCppBytes c;
    // c.SetData(bytes);  // I didn't set the data here

    c.GetData().get();
    c.GetData()->GetSize();
    c.GetData()->GetBuffer();
}

I think will got someting like crash in wamr or wasm. But it was finished normally.

Then I tested the similar codes in android ndk, and I got nullptr crash:
A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
A/DEBUG: Cause: null pointer dereference

So why?

xujuntwt95329, 2021-03-13:
As a managed runtime, WAMR shouldn‘t crash even though the application access some invalid memory space. For example, if you write `int c = *(int *)(-100)` in C, then WAMR will throw `Exception: out of bounds memory access`, but it will not crash.

As for your case, the runtime doesn't throw out the OOM exception because `nullptr` in C++, which is actually a number `0`, is a valid memory space in WebAssembly, WAMR will not avoid you from accessing this area, but this will cause an undefined behaviour.

xingkaiyu, 2021-03-15:
@xujuntwt95329 
We should avoid this kind of operation from source code level.
—— I can agree more and this is why I ask this question. If I got a error here, then I will find the code bug easier.
—— Thanks for your answer. I will note it in my code.

closed 2021-03-15 by xingkaiyu

xujuntwt95329, 2021-03-15:
Yes, an error will make it easier to find out the issue, but we can't really rely on this kind of errors, because dereferencing a nullptr in C++ is undefined behavior, different compilers/platforms may introduce different behavior