wasm-micro-runtime/wasm-micro-runtime #572: Different string has same address problem

xingkaiyu, 2021-03-13:
I wrote a application to test std::string by using wamr and wasi-sdk-12.0.
When I wrote the codes like below, I got the problem.

First, I wrote a class which has 2 private string members, getters and setters:
 class ClassString {
private:
    std::string s1;
    std::string s2;

public:
    std::string GetS1() {
        return s1;
    }

    std::string GetS2() {
        return s2;
    }

    void SetS1(const std::string &s) {
        s1 = s;
    }

    void SetS2(const std::string &s) {
        s2 = s;
    }
};

Then I wrote a struct has to const char * members:
struct StructString {
    const char *c1;
    const char *c2;
};

Then I wrote the test function like this and test it:
void TestStringPtr()
{
    ClassString cl;
    cl.SetS1("123");
    cl.SetS2("456");

    StructString st;
    st.c1 = cl.GetS1().c_str();
    st.c2 = cl.GetS2().c_str();
    LOGI("TestStringPtr st.c1 is %p st.c2 is %p st.c1 is %s st.c2 is %s", 
        st.c1, st.c2, st.c1, st.c2);
}

And then I got the log below:
TestStringPtr st.c1 is 0x11598 st.c2 is 0x11598 st.c1 is 456 st.c2 is 456
The 2 char * member are same after been Assigned

But when I try the same code in android ndk, I got the different log:
TestStringPtr st.c1 is 0x7fef96e441 st.c2 is 0x7fef96e429 st.c1 is 123 st.c2 is 456

I have tryed the ClassString getters return string & instead of string in wasm app, and I got two different char*s.

I guess may be wamr think the temporary address made by st.c1 = cl.GetS1().c_str(); is not useful after the assignment and released it. And when st.c2 = cl.GetS2().c_str(); , the same address is used again.

Is this a problem or wamr do it deliberately?

xujuntwt95329, 2021-03-13:
> I guess may be wamr think the temporary address made by st.c1 = cl.GetS1().c_str(); is not useful after the assignment and released it. And when st.c2 = cl.GetS2().c_str(); , the same address is used again.

WAMR never does this kind of operation, WebAssembly is an instruction-level VM, it just executes the given bytecode one by one, and WAMR doesn't have any information about your application-level semantics.

> Is this a problem or wamr do it deliberately?

I believe this should be a problem of `wasi-sdk-12`, I build your sample with `wasi-sdk-11` and it works well.

xingkaiyu, 2021-03-16:
@xujuntwt95329 
Thanks for your answer.
I think this problem meybe is my mistake. cl.GetS2() returned a temporary std::string, but I didn't put it into a string var and only get c_str. c++ may think to temporary std::string is no longer used and free it. Then the second cl.GetS2() used the released address to put it temporary std::string.
I have changed it getters return const std::string &, and it works.

closed 2021-03-16 by xingkaiyu