wasm-micro-runtime/wasm-micro-runtime #575: Is wasm module must be reload when call wasm_runtime_call_wasm get exception

xingkaiyu, 2021-03-16:
Hi~
I used wamr and sometimes I will get runtime exception when called wasm_runtime_call_wasm (mostly it is my code bug). 
Such as "failed to call unlinked import function (env, XXXX)", "Exception: unreachable", "out of bounds memory access".
And I found that when I got any exception, I will not get success call anymore (these calls‘ code is right), unless I reload the wasm from get .wasm file (the runtime need not be reload)
So, is wamr always need to be reload when get false from wasm_runtime_call_wasm?

lum1n0us, 2021-03-16:
Yeah, I am afraid so.

WAMR will quit ASAP when meeting any exception.

xujuntwt95329, 2021-03-16:
You can use `wasm_runtime_set_exception(module_inst, NULL)` to clear the exception and then you can call your other functions.

xingkaiyu, 2021-03-24:
It is ok now. Thanks ~

closed 2021-03-24 by xingkaiyu