wasm-micro-runtime/wasm-micro-runtime #585: Android WASI support

JavanZhu, 2021-03-22:
Hello, I am trying to build wasm with WASI for android platform and got following build error:
``` 
ld.lld: error: undefined symbol: preadv
ld.lld: error: undefined symbol: pwritev
ld.lld: error: undefined symbol: posix_fadvise
ld.lld: error: undefined symbol: posix_fallocate
ld.lld: error: undefined symbol: linkat
ld.lld: error: undefined symbol: symlinkat
ld.lld: error: undefined symbol: readlinkat
ld.lld: error: undefined symbol: telldir
ld.lld: error: undefined symbol: seekdir
ld.lld: error: undefined symbol: futimens
ld.lld: error: undefined symbol: getrandom

```
and found that there have an [experiment commit about it](https://github.com/xujuntwt95329/wasm-micro-runtime/commit/f6d1d4be1af55e09b524355c1c25931d8de749b8) by @XuJun2019 (Thanks) after searched from [issues](https://github.com/bytecodealliance/wasm-micro-runtime/issues/520). 

Compiler will throw above error if we don't set ANDROID_PLATFORM ( the default value is "android-16") in cmake, Because Android system doesn't support those API. But not all android platform doesn't support this API, I list the compatible list of above API in Android platform.

API name | mini __ANDROID_API__
-- | --
preadv | 24

pwritev | 24
posix_fadvise | 21
posix_fallocate | 21
linkat | 21
symlinkat | 21
readlinkat | 21
telldir | 23
seekdir | 23
futimens | 19
getrandom | 28

So maybe we could improve this [experiment commit](https://github.com/xujuntwt95329/wasm-micro-runtime/commit/f6d1d4be1af55e09b524355c1c25931d8de749b8) by checking ANDROID_API.
I upload a [patch](https://github.com/bytecodealliance/wasm-micro-runtime/files/6180183/android_wasi_support.patch.zip) about it. Please tell me if there have any problem. If needed I can make a pull request about it and test in my android device, Thanks.

JavanZhu, 2021-03-22:
Sorry, previous patch has problem in darwin, and I have updated the patch file.

wenyongh, 2021-03-23:
Hi, @JavanZhu, it would be great if Libc WASI is supported on Android platform, I read the patch and have some comments:
1. Had better not modify source code of libc-wasi, modify source code in android platform instead. For the patch, the changes to ssp_config.h is OK, but the changes of posix.c can be moved to file platform_internal.h and platform_init.c under core/shared/platform/android folder: in platform_internal.h, declares the function prototypes and data type (e.g., __syscall_slong_t), and in platform_init.c, if the function isn't implemented, define it, leave empty and return fail (most functions should return -1 but not 0), e.g.
```C
//platform_internal.h
int posix_fadvise(int fd, off_t offset, off_t len, int advice);

//platform_init.c:
#if __ANDROID_API__ < 21
int posix_fadvise(int fd, off_t offset, off_t len, int advice)
{
  return -1;
}
#endif
```
2. There is no macro of android version to control some of the functions, e.g. symlinkat and telldir, please add it.
3. Had better create a PR, so that we can add comments and make changes.

closed 2021-03-24 by JavanZhu