wasm-micro-runtime/wasm-micro-runtime #593: windows and android cmake suggestion

xingkaiyu, 2021-03-29:
Windows:
1- I think the line 94 in windows cmake file is wrong.  
    Now it is: 
    set (CMAKE_SHARED_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} /SAFESEH:NO")
    I think the right one is 
    set (CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} /SAFESEH:NO")
2- Now in wasm_export.h we have a macro logic using MSVC to make difference when use windows.
     But in windows cmake file. we have not set the MSVC to 1.
     I think it is better to set it in cmake

Android:
1- I see you have add ${UV_A_LIBS} in linux and windows cmake. Can you also add it in android target_link_libraries commond
2- I see you have add set (ANDROID_PLATFORM 24) in Android cmake. But I think it may should be set (ANDROID_PLATFORM android-24). Please check.

lum1n0us, 2021-03-31:
An answer for question 2 of Android:

Refer to Toolchain arguments of CMAKE: https://developer.android.com/ndk/guides/cmake#android_platform

Multiple formats are accepted for this parameter (`ANDROID_PLATFORM`) :
- *android-$API_LEVEL*
- *$API_LEVEL*
- *android-$API_LETTER*

xingkaiyu, 2021-03-31:
@lum1n0us 
thanks for your guide~
And what about others?

wenyongh, 2021-04-01:
Hi,

> Windows:
> 1- I think the line 94 in windows cmake file is wrong.
> Now it is:
> set (CMAKE_SHARED_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} /SAFESEH:NO")
> I think the right one is
> set (CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} /SAFESEH:NO")

Fixed, not sure whether uvwasi works in android platform, you can have a try.

> 2- Now in wasm_export.h we have a macro logic using MSVC to make difference when use windows.
> But in windows cmake file. we have not set the MSVC to 1.
> I think it is better to set it in cmake

Need more time to confirm it.

> 
> Android:
> 1- I see you have add ${UV_A_LIBS} in linux and windows cmake. Can you also add it in android target_link_libraries commond

Fixed.

> 2- I see you have add set (ANDROID_PLATFORM 24) in Android cmake. But I think it may should be set (ANDROID_PLATFORM android-24). Please check.

xingkaiyu, 2021-04-02:
@wenyongh  
I see you changed Windows-1 and Android-1 in current code. Thanks.
And I don't know what means by "Need more time to confirm it". Is it means that you need to check if set MSVC to 1 in windows cmake is right?

For "not sure whether uvwasi works in android platform, you can have a try."
We are using uvwasi in android now. Compare to origin wasi implement, uvwasi can be compiled to lower android version (arm64-v8a at least to 19 and armeabi-v7a to 21). And it seems that uvwasi can do more, like using sqlite.

wenyongh, 2021-04-02:
Hi @xingkaiyu, for macro MSVC in Windows platform, we need to check whether the macro is pre-defined by Visual Studio, or we should define it in makefile. And check whether it impacts the compilation process for Windows platform and other platforms.

xingkaiyu, 2021-04-08:
@wenyongh  ok. thanks

closed 2021-04-08 by xingkaiyu

wenyongh, 2021-04-09:
The MSVC macro issue was fixed in #604, could you please try again? Thanks.