wasm-micro-runtime/wasm-micro-runtime #594: std::string using problem when enable pthread in android

xingkaiyu, 2021-03-30:
Hi~
I had write an issue about string using before. At that time, I think it is because I am using wasi-sdk-12.0. 
But today, I tryed more and got some new idea.
The updated conditions are:
1- enable pthread both in wasm app and wamr (but I didn't use thread in my code)
2- use wasi-sdk-12.0 to build the wasm app

And my code changed more simple (TestWasi12String is the entry function):
class TestWasi12StringClass {
public:
    TestWasi12StringClass(std::string uuid)
    {
        mUuid = std::string(uuid);
        LOGI("TestWasi12StringClass mUuid %s uuid %s", mUuid.c_str(), uuid.c_str());
    }

public:
    std::string mUuid;
};

std::string Test()
{
    const char *cstr_native = "a string in wasm";
    // const char *cstr_native = GetStringNative();
    std::string str(cstr_native);
    LOGI("str is %s", str.c_str());
    TestWasi12StringClass class1(str);
    return str;
}

void TestWasi12String()
{
    std::string test = Test();
    LOGI("TestWasi12String test is %s", test.c_str());
}

LOGI is a function to use vsprintf to format log and call native android log to write log:
#define TAG "TestPthreadRandom"
void LOGI(const char *fmt, ...)
{
    char msg[1024];
    va_list va;
    va_start(va, fmt);
    vsprintf(msg, fmt, va);
    va_end(va);

    LogInfoNative(TAG, msg);
}

GetStringNative is a native function only return a c_str's wasm address to wasm app:
uint32_t GetStringNative(wasm_exec_env_t exec_env)
{
    const char *result = "a string from native";
    wasm_module_inst_t moduleInst = wasm_runtime_get_module_inst(exec_env);
    uint32_t ret = wasm_runtime_module_dup_data(moduleInst, result, strlen(result) + 1);
    LogInfo(TAG, "GetStringNative result is %d", ret);
    return ret;
}

When I use "const char *cstr_native = "a string in wasm";", I can get the normal result like:
I/fwkit_wasm_TestPthreadRandom: str is a string in wasm
I/fwkit_wasm_TestPthreadRandom: TestWasi12StringClass mUuid a string in wasm uuid a string in wasm
I/fwkit_wasm_TestPthreadRandom: TestWasi12String test is a string in wasm

But when I use "const char *cstr_native = GetStringNative();", I will get wrong log:
I/fwkit_wasm_TestPthreadRandom: str is a string from native
I/fwkit_wasm_TestPthreadRandom: TestWasi12StringClass mUuid   uuid �
I/fwkit_wasm_TestPthreadRandom: TestWasi12String test is  
We can see the first string is normal, but the others are garbled.

I have tryed some little change in my code or cmake file (all use GetStringNative()):
1- remove all the pthread param in cmake file (get right log):
    removed all below (only remove -pthread still get wrong log):
    # set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} \
    #     -pthread \
    #     -Wl,--shared-memory,--max-memory=2097152 \
    #     -Wl,--no-check-features \
    #     -Wl,--export=__heap_base,--export=__data_end \
    #     -Wl,--export=__wasm_call_ctors \
    #     -Wl,--export=malloc,--export=free \
    #     ")
2- add malloc and memcpy to copy the native result (also only str is right):
std::string Test()
{
    // const char *cstr_native = "a string in wasm";
    const char *cstr_native = GetStringNative();
    char *cstr_wasm = (char *)malloc(strlen(cstr_native) + 1);
    memcpy(cstr_wasm, cstr_native, strlen(cstr_native) + 1);
    std::string str(cstr_wasm);
    LOGI("str1 is %s", str.c_str());
    TestWasi12StringClass class1(str);
    return str;
}

xujuntwt95329, 2021-04-01:
Hi, thanks for reporting this issue. When using `wasm_module_dup_data` API with lib-pthread enabled, the aux stack space may be covered, so we encounter some unexpected behavior.

This bug has been fixed in PR #596 , could you please try again ?

xingkaiyu, 2021-04-02:
thanks. I think it is ok now

closed 2021-04-02 by xingkaiyu