wasm-micro-runtime/wasm-micro-runtime #595: Crash when execute aot file

JavanZhu, 2021-03-31:
Hi, I compiled the wamrc from latest wamr code, and use it to convert a wasm file. When execute aot file on wamr, it crashes with error " signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr" on android or "Exception: EXC_BAD_ACCESS (code=2, address=0x10553cac4)" on darwin. 
cmake output:
```
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build as target X86_64
-- CMAKE_BUILD_TYPE = Release
-- Found LLVM 11.1.0
-- Using LLVMConfig.cmake in: wasm-micro-runtime/core/deps/llvm/build/lib/cmake/llvm
-- The ASM compiler identification is Clang
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Configuring done
-- Generating done
-- Build files have been written to: wasm-micro-runtime/wamr-compiler/build

```
```
./wamrc --disable-simd --target=x86_64 --size-level=1 -o hello_world.aot hello_world.wasm
./iwasm -f onLoad hello_world.aot
```
```
unhandled SIGBUS, si_addr: 0x10ec3b994
[1]    66121 abort      ./iwasm -f onLoad 
```
the crash position of source code:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/5a13e1bbbccf2a2db9f256aa948bfde2af1aa946/core/iwasm/aot/aot_runtime.c#L791

* platform: android / darwin

I made a minimal example to reproduce it [crash_example.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6236676/crash_example.zip), could you please have a try. Hope above information is useful, please tell me if you need more information about it, thank you.

wenyongh, 2021-04-01:
Hi, I tested the AOT file in attachment provided by you and the AOT file generated with the wamrc command provided by you, both work well in linux platform. As it is not easy to set up the android/darwin platform quickly, could you please pull the latest code, rebuild wamrc and iwasm, and try again? If the issue still exists, we will try to setup android/darwin environment and test it.

JavanZhu, 2021-04-01:
> Hi, I tested the AOT file in attachment provided by you and the AOT file generated with the wamrc command provided by you, both work well in linux platform. As it is not easy to set up the [...]

Ok, I 'll try it again. BTW, I noticed when I updated wamr code, I need to recompile the wamrc from latest code, and then use the new version wamrc to recompile the wasm to aot file.(If I didn't do it like this, the new runtime will crash when it load the old aot file). So which part of the code modification in wamr caused us to do it like this? (not sure I understand it right), Thanks.

JavanZhu, 2021-04-01:
Hi @wenyongh, after this [commit](https://github.com/bytecodealliance/wasm-micro-runtime/commit/68ff3a830e6afba396a303ae416704055dd69b94), iwasm will crash on Darwin. And It works well if I comment following lines. Hope this information is useful, Thanks.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/7db2221ad94cd73e88e958a2040e2b66a75519e1/core/iwasm/aot/aot_loader.c#L1840-L1869

wenyongh, 2021-04-01:
Hi, I uploaded a patch, could you please try again? [aot_loader.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6243545/aot_loader.zip)

JavanZhu, 2021-04-01:
HI @wenyongh, this patch is works, thanks.

closed 2021-04-01 by JavanZhu

JavanZhu, 2021-04-01:
Hi @wenyongh, Thank you for your help, and may I ask about a question about why this problem is only exist in darwin/android platform?(I never test it in other platforms), thanks.

wenyongh, 2021-04-02:
Hi, the AOT code should be set not only readonly, but also executable. I am not sure why the issue only exists in darwin/android, normally it should also happen in linux/windows platform, but it runs fine on them.