wasm-micro-runtime/wasm-micro-runtime #598: Question about aot mode

JavanZhu, 2021-04-02:
Hi, my wasm app is running on the old version of wamr ([this commit](https://github.com/bytecodealliance/wasm-micro-runtime/commit/2dd3875fd909bbfa2e041e14f4ed2155d5b5f80b)), now I'm going to pull the latest code of wamrc and rebuilding it. But when the new version iwasm loaded the old version aot file, I found that it will throw "Exception: out of bounds memory access" in function wasm_runtime_instantiate(). And rebuilding the wamrc on the latest code of wamr, then using it to rebuild a wasm app file again to generate a new version aot file can solve this problem. So I have three question about this:

1. which part of code changes in the wamr project could cause this problem?
2. could you please tell me some detail information about the reason about this error?
3. Now if we want to run aot file on iwasm, we need to generate aot file on computer, then we push the aot file to
the product device. So is it possible to change above process to the following? 
```
wasm file -> iot device -> if it want to run on aot mode and aot file is new-> build in device(not computer) -> generate aot file -> load aot file
(If we were changed the aot file generation process from computer to product device, Can aot mode be better cross-platform?
(seems that android ART runtime is doing like this))
```
(Because my customer's devices have all kind of cpu architecture and platform, if I want to upgrade the iwasm which was running on the customer's device, I have to prepare a corresponding aot file (new version). It's will become more and more difficult when our devices become more and more, And that doesn’t even consider the user’s unwillingness to upgrade)
So I am not sure this could be achieved. And any information or tips about it is welcome and I would appreciate about it, thanks.

wenyongh, 2021-04-10:
Hi, the aot code and runtime work together to access to the aot instance data, the aot file format and the layout of some runtime data structures are pre-defined, meaning that runtime knows the file format of aot file and aot code knows the layout of some runtime data structures, which normally include AOTModuleInstance, AOTMemoryInstance and WASMExecEnv. Though we design the aot file format and the runtime data structures very carefully, and define AOT version(see AOT_CURRENT_VERSION in core/config.h, current version is 2) to add check when loading AOT file, it is really not easy to ensure that the older AOT file can be correctly run by latest runtime as there might be some new spec features added.

> 1. which part of code changes in the wamr project could cause this problem?
> 2. could you please tell me some detail information about the reason about this error?

I roughly checked the changes from the commit you mentioned, it seems that the change of AOTMemoryInstance structure introduced by SIMD feature causes the issue: field `mem_bound_check_16bytes` is added, the layout of AOTMemoryInstance isn't consist with the layout pre-defined for AOT code, you might comment the field and build iwasm again to have a try.
 
> 3. Now if we want to run aot file on iwasm, we need to generate aot file on computer, then we push the aot file to
>    the product device. So is it possible to change above process to the following? [...]

Yes, that seems a good way, not sure whether it is easy to port wamrc to your device, the footprint is large. If it is OK, had better just build one target required, e.g., change the line `-DLLVM_TARGETS_TO_BUILD:STRING="..."` in build_llvm.sh. But how to check whether aot file new and need to build it in device?

JavanZhu, 2021-04-13:
Hi @wenyongh, Thank you very much for your reply, It's very helpful. And after Commenting the `mem_bound_check_16bytes` field you mentioned above, it solves the error "Exception: out of bounds memory access", and the hello world aot file loads successfully. But my wasm app is complex than this, and I found it will crash in another problem. 
the crash stack is:
```
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x15
Cause: null pointer dereferenc
...
#00 wasm_runtime_invoke_native()
#01 aot_call_indirect()
#02 pc 0001d38e  <anonymous:e2816000>
```
https://github.com/bytecodealliance/wasm-micro-runtime/blob/ee97618bdb1ac5b54f6423a921a4741e1a9e4358/core/iwasm/common/wasm_runtime_common.c#L3301
[image]

And for question `how to check whether aot file new and need to build it in device?`, If wamr runtime could export an api like `build_aot_file(origin_wasm_file_path, aot_file_path, build_config)` for developer to call,  when they install new runtime to their devices, they can rebuild all exist wasm files to regenerate the new aot files, maybe it's easy for developer to check whether aot file new.

wenyongh, 2021-04-13:
Hi @JavanZhu, it seems that the crash issue is caused by the change of function prototype of aot_call_indirect, the arguments `bool check_func_type, uint32 func_type_idx` were removed in this [commit](https://github.com/bytecodealliance/wasm-micro-runtime/commit/788cbf2a1945cf04cc8139f7a363e268f18cbe34). You might restore the code of aot_call_indirect function to have a try.

For the aot file upgrade, it's not suitable for runtime to export the build_aot_file api, in fact it should be easy to implement it in your host native or runtime embedder (e.g. system("wamrc -o xxx.aot xxx.wasm")) and check whether to upgrade it before loading AOT file (e.g. whether wamrc or iwasm is upgraded), could you please implement it by yourself? Thanks.

JavanZhu, 2021-04-14:
Ok, it works, Thank you very much for your help!

closed 2021-04-14 by JavanZhu