wasm-micro-runtime/wasm-micro-runtime #600: windows x86_64 load aot failed

bxkai, 2021-04-02:
hi, On windows x86_64, load aot failed: invalid relocation type 11. 
1）Build [helloword](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/app-samples/hello-world/main.c) wasm app 
/opt/wasi-sdk/bin/clang -O3 \
        -z stack-size=4096 -Wl,--initial-memory=65536 \
        -o test.wasm main.c \
        -Wl,--export=main -Wl,--export=__main_argc_argv \
        -Wl,--export=__data_end -Wl,--export=__heap_base \
        -Wl,--strip-all,--no-entry \
        -Wl,--allow-undefined 
2）wamrc --target=x86_64 --disable-simd -o test_std_64.aot test.wasm
      wamrc --target=x86_64 --cpu=corei7 -o test_std_64_core.aot test.wasm
3)  iwasm.exe test_std_64.aot
Load relocation section failed: invalid relocation type 11.
    iwasm.exe test_std_64_core.aot
Load relocation section failed: invalid relocation type 11.

But build wasm app with -nostdlib , aot load success

wenyongh, 2021-04-09:
@bxkai thanks for reporting the issue! We have fixed it in #606, could you please try again? Thanks.

bxkai, 2021-04-12:
@wenyongh aot load success when wamrc with --size-level=1 option . or not
AOT module load failed: relocation truncated to fit IMAGE_REL_AMD64_ADDR32 failed. Try using wamrc with --size-level=1 option.

wenyongh, 2021-04-13:
@bxkai yes, please add `--size-level=1` for wamrc on Windows now.

closed 2021-04-19 by bxkai