wasm-micro-runtime/wasm-micro-runtime #602: Pthread stability problem

xingkaiyu, 2021-04-08:
[code.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/6275606/code.zip)

Hi~
As mentioned before, I have write the code as attach zip file.
The wasm app folder is the whole code of wasm app, including cmake file
The native folder are the main codes in native, not including wasm_runtime_init and wasm_runtime_register_natives. But sure I do it in other place.

My code is a test for pthread doing cjson parse, both in main and sub thread.
wamr:
1- I am using 03-25 release and add the patch https://github.com/bytecodealliance/wasm-micro-runtime/pull/596
2- I am using uvwasi
3- I set pthread = 1 always
wasm app:
1- Bool param "useThread" is used to make the json parse do in main or sub thread. 
2- String param "jsonParam" is json string from native to be parse here
3- create_monitor_with_helpers using cjson to make a json result from native
native:
1- the entry function is Java_com_example_wamrtest_pthread_PthreadOtherActivity_testCJsonExampleInNative in native-lib-pthread-random.cpp
2- I am using android, so I use AssetManager to get wasm app file
3- Every time I load wasm app file, module, module_inst, exec_env first. Then do wasm app function 100 times. And unload wasm app file, module, module_inst, exec_env in the end
4- Native also has 2 modes: call wasm app in current thead or in sub thread.

And the result is:
1- When I compile wasm app without pthread param (comment the set in cmake under "set when use thread"), I can always get right result, no matter I call wasm in current or sub thread in naitve (of cause I can only use main thread in wasm app).
2- When I compile wasm app with pthread param
    2-1 when I call wasm in current thread in native
          when wasm app run without use thread, it is stable
          when wasm app run in sub thread, the string param will be garbled with high probability and I will get "out of bounds memory access" exception. And I even can not enter wasm app function and get "out of bounds memory access" exception 
immediately in end.
    2-2 when I call wasm in sub thread in native
           No matter I use sub thread or not in wasm app, the results are same. In the beginning,  I will get "native stack overflow" exception and the string param I get is null in most cases. And I get "out of bounds memory access" in the end and can enter wasm function.

Would you please check this, thanks~

xujuntwt95329, 2021-04-09:
Hi, this issue is caused by wasi-libc's `dlmalloc`, currently the implementation of `dlmalloc` in wasi-sdk is not thread safe (it's not an atomic operation), so when you call malloc in multiple threads, the heap may be destroyed.

As a workaround, you can hack the wasi-sdk's libc, remove the `dlmalloc` related functions and use WAMR's builtin memory allocator.

1. hack the wasi-libc
``` bash
# don't forget to backup the original files
sudo /opt/wasi-sdk/bin/llvm-ar -d /opt/wasi-sdk/share/wasi-sysroot/lib/wasm32-wasi/libc.a dlmalloc.o
```

2. modify cmake file:
remove `-Wl,--export=malloc,--export=free`, and add `malloc`, `free`, `calloc` and `realloc` to the `defined-symbols.txt`

3. add wrapper for realloc
Currently WAMR doesn't provide realloc wrapper, but it is referenced in cJSON library, so you have to add a wrapper for it:

in `wasm_runtime_common.c`:
``` C
uint32
wasm_runtime_module_realloc(WASMModuleInstanceCommon *module_inst, uint32 ptr,
                            uint32 size, void **p_native_addr)
{
#if WASM_ENABLE_INTERP != 0
    if (module_inst->module_type == Wasm_Module_Bytecode)
        return wasm_module_realloc((WASMModuleInstance*)module_inst, ptr,
                                   size, p_native_addr);
#endif
#if WASM_ENABLE_AOT != 0
    if (module_inst->module_type == Wasm_Module_AoT)
        // return aot_module_malloc((AOTModuleInstance*)module_inst, size,
        //                          p_native_addr);
        return 0;
#endif
    return 0;
}
```

in `wasm_runtime_common.h`:
``` C
uint32
wasm_runtime_module_realloc(WASMModuleInstanceCommon *module_inst, uint32 ptr,
                            uint32 size, void **p_native_addr);
```

in `wasm_runtime.c`:
``` C
uint32
wasm_module_realloc(WASMModuleInstance *module_inst, uint32 ptr, uint32 size,
                    void **p_native_addr)
{
    WASMMemoryInstance *memory = module_inst->default_memory;
    uint8 *addr = NULL;

    if (!memory) {
        wasm_set_exception(module_inst, "uninitialized memory");
        return 0;
    }

    if (memory->heap_handle) {
        addr = mem_allocator_realloc(memory->heap_handle,
                                     memory->memory_data + ptr, size);
    }
    if (!addr) {
        wasm_set_exception(module_inst, "out of memory");
        return 0;
    }
    if (p_native_addr)
        *p_native_addr = addr;

    return (uint32)(addr - memory->memory_data);
}
```

in `libc_builtin_wrapper.c`:
``` C
static uint32
realloc_wrapper(wasm_exec_env_t exec_env, uint32 orig_addr, uint32 new_size)
{
    void *new_addr = NULL;
    wasm_module_inst_t module_inst = get_module_inst(exec_env);

    return wasm_runtime_module_realloc(module_inst, orig_addr, new_size, NULL);
}
```
and add the wrapper to the list:
``` C
static NativeSymbol native_symbols_libc_builtin[] = {
    // ......
    REG_NATIVE_FUNC(malloc, "(i)i"),
    REG_NATIVE_FUNC(realloc, "(ii)i"),     // <------ add this line
    REG_NATIVE_FUNC(calloc, "(ii)i"),
    // ......
}
```

After doing these things, your case should work correctly in thread mode.

xujuntwt95329, 2021-04-09:
The `realloc` wrapper has been implemented in PR #605 
Now you only need step 1 and 2 mentioned above.

xingkaiyu, 2021-04-10:
@xujuntwt95329  
Thanks, I think the problem is mostly solved by your suggestion.
But I still have some questions:
1- I got crash when call wasm_application_execute_main after all the change.
    Call Ctors and Dtors are ok, so I called them instead.
2- I will got compile error when using both nlohmann::json and -pthread in wasm app
    the error is: fatal error: error in backend: only -ftls-model=local-exec is supported for now on non-Emscripten OSes: variable errno
    I am using wasi-12.0
    nlohmann::json is from https://github.com/nlohmann/json/tree/develop/single_include/nlohmann you can only include json.hpp in your code and add one line like this: std::string retJson = nlohmann::json::parse("");
   Complie without -pthread is ok and I think the wasm works well
3- I did some more test, and got some new error ("invalid argument count 0, expected 2"). But my code this time is too complex. I will try if it is my falt or if I can make the code simpler...

xujuntwt95329, 2021-04-12:
> 1- I got crash when call wasm_application_execute_main after all the change.

`wasm_application_execute_main` will destroy the `exec_env` immediately once your main thread exit, so it's not suitable for your scenario. 
However, the runtime shouldn't crash even though it's not the correct usage. I've tried to use `wasm_application_execute_main` in your case, but I didn't encounter a crash, could you please update your modified code?

> fatal error: error in backend: only -ftls-model=local-exec is supported for now on non-Emscripten OSes: variable errno

This is a known limitation mentioned in [pthread_library.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md#known-limits). The program may work well without `-pthread` option, but the 
behaviour of `errno` may be unexpected.

> I did some more test, and got some new error ("invalid argument count 0, expected 2").

WAMR detects that you want to call a function with wrong arguments count, so this exception is thrown. It's not an error during execution, you should check the `argc` parameter passed to `wasm_runtime_call_wasm`

xingkaiyu, 2021-04-13:
@xujuntwt95329  about the 3rd question, I did same more test. The result is:
The real error info is not "invalid argument count 0, expected 2" but is "app heap corrupted" from wasm_module_malloc when I called it to send buffer(string) to wasm. I got "invalid argument count 0, expected 2" is because when malloc failed I will pass 0 param to wasm.

And then I do some test by add log to wamr and I found that the error is caused by remove_tree_node in ems_alloc.c. 
Sometimes I will got a out of bound parent addr and hmu_is_in_heap(parent, base_addr, end_addr) return false, so the is_heap_corrupted will be set to true. And then it will cause wasm_module_malloc failed.

My log when error is like this:
I/wasm_runtime::: xingkaiyu-error remove_tree_node p is 0x702f61e4bc
I/wasm_runtime::: xingkaiyu-error remove_tree_node parent is 0x65646f635f737574
I/wasm_runtime::: xingkaiyu-error remove_tree_node root is 0x702fc7ac40
I/wasm_runtime::: xingkaiyu-error remove_tree_node base_addr is 0x702f61ce04
I/wasm_runtime::: xingkaiyu-error remove_tree_node end_addr is 0x702fa1cdfc

I am not sure why this happen and I will do more tests. And I hope you may give me some suggestion~

xingkaiyu, 2021-04-15:
@xujuntwt95329  
I the problem before I am sure now it was my mistake that I did not use malloc point when thrans param to sub thread...
But I got a new question about the patch https://github.com/bytecodealliance/wasm-micro-runtime/pull/596
I think the aot_create_exec_env_and_call_function or wasm_create_exec_env_and_call_function should always call wasm_exec_env_set_thread_info after get exec_env no mater from cluster or by create to refress the handler and thread stack start adress. 
But now wasm_exec_env_set_thread_info will not be called when find env from cluster and some time it will cause "native stack overflow" in wasm_interp_call_wasm because the native thread changed. 
Could you please check it~

xujuntwt95329, 2021-04-19:
> But now wasm_exec_env_set_thread_info will not be called when find env from cluster and some time it will cause "native stack overflow" in wasm_interp_call_wasm because the native thread changed.

I've checked the code and this should not be the problem. If we can get `exec_env` from the `cluster`, then it means the `exec_env` has already been bound to the current thread.

I think there may be some other reasons that cause the `native stack overflow` exception, for example, do you create new `exec_env` in your native API wrappers?

xingkaiyu, 2021-04-19:
@xujuntwt95329  
I have not built a new exec_env. I will create a env, hold it and reuse it until the whole android app is shut down.
But I guess, it is why I may got the problem.
As you said, the exec_env will be bound to the thread and then:
1- first, I create a env and use it to do something like do malloc in wasm. At this time the env is bound to the first thread (the thread is created by android)
2- then, I start a new call from android. This time, android will use a new thread and because the process is same, I still use the env created in the first step.
3- I call malloc function (wasm_runtime_module_dup_data) before the wasm_runtime_call_wasm because I need to pass some buffer to wasm app
4- then the chain is wasm_runtime_module_dup_data->wasm_module_dup_data->wasm_module_malloc->execute_malloc_function->wasm_create_exec_env_and_call_function and there is no place to set the relation of the new thread and the env

xingkaiyu, 2021-04-20:
I add some log in both natvie and wamr. I think it can help you to check the problem:

1- native created a exec_env(0x6ee2800000) for module_inst(0x6f4894c480) when init. this time native_stack_boundary is not set yet(0x0).
    _2021-04-20 10:57:15.094 28264-28264 I/wamr-manager: xingkaiyu-test android native call wasm_runtime_create_exec_env module_inst is 0x6f4894c480
    2021-04-20 10:57:15.094 28264-28264 I/wasm_runtime::: xingkaiyu-test from wasm_runtime_create_exec_env module_inst is 0x6f4894c480
    2021-04-20 10:57:15.096 28264-28264 I/wasm_runtime::: xingkaiyu-test wasm_exec_env_create module_inst is 0x6f4894c480 exec_env is 0x6ee2800000 exec_env->native_stack_boundary is 0x0
    2021-04-20 10:57:15.096 28264-28264 I/wamr-manager: xingkaiyu-test android native wasm_runtime_create_exec_env finished exec_env is 0x6ee2800000_

2- native call __wasm_call_ctors in android main thread, using exec_env(0x6ee2800000). 
   and by using wasm_runtime_call_wasm to call __wasm_call_ctors, native_stack_boundary is set to 0x7ff708d200 associated with thread-28264-28264(480612794064)
   And it called wasm_runtime_module_dup_data (I guess) to malloc some space for globle variable. This time, the exec_env->native_stack_boundary is right, calling well.
    _2021-04-20 10:57:15.096 28264-28264 I/wamr-manager: xingkaiyu-test android native call __wasm_call_ctors exec_env is 0x6ee2800000
    2021-04-20 10:57:15.106 28264-28264 I/wasm_runtime::: xingkaiyu-test wasm_runtime_call_wasm done wasm_exec_env_set_thread_info exec_env is 0x6ee2800000 exec_env->native_stack_boundary is 0x7ff708d200
    2021-04-20 10:57:15.107 28264-28264 I/wasm_runtime::: xingkaiyu-test from wasm_runtime_module_dup_data module_inst is 0x6f4894c480
    2021-04-20 10:57:15.107 28264-28264 I/wasm_runtime::: xingkaiyu-test from wasm_module_malloc execute_malloc_function module_inst is 0x6f4894c480
    2021-04-20 10:57:15.107 28264-28264 I/wasm_runtime::: xingkaiyu-test wasm_create_exec_env_and_call_function now pthread_self is 480612794064 module_inst is 0x6f4894c480 exec_env is 0x6ee2800000 exec_env->native_stack_boundary is 0x7ff708d200
    2021-04-20 10:57:15.108 28264-28264 I/wamr-manager: xingkaiyu-test android native __wasm_call_ctors finished exec_env is 0x6ee2800000_

3- native called a normal function in android sub thread, using exec_env(0x6ee2800000). 
   And before calling, wasm_runtime_module_dup_data was called first to transfer some buffer to wasm app.
   Because the exec_env is found from the cluster, the wasm_exec_env_set_thread_info is not called.
   But the native thread is changed, so as the native_stack_boundary. And then, I got native stack overflow exception.
    _2021-04-20 10:57:15.485 28264-28431 I/wamr-manager: xingkaiyu-test android native normal function before, call wasm_runtime_module_dup_data module_inst is 0x6f4894c480 exec_env is 0x6ee2800000
    2021-04-20 10:57:15.485 28264-28431 I/wasm_runtime::: xingkaiyu-test from wasm_runtime_module_dup_data module_inst is 0x6f4894c480
    2021-04-20 10:57:15.485 28264-28431 I/wasm_runtime::: xingkaiyu-test from wasm_module_malloc execute_malloc_function module_inst is 0x6f4894c480
    2021-04-20 10:57:15.485 28264-28431 I/wasm_runtime::: xingkaiyu-test wasm_create_exec_env_and_call_function now pthread_self is 476229647696 module_inst is 0x6f4894c480 exec_env is 0x6ee2800000 exec_env->native_stack_boundary is 0x7ff708d200
    2021-04-20 10:57:15.485 28264-28431 I/wasm_runtime::: xingkaiyu-test wasm_interp_call_wasm stack overflow. exec_env is 0x6ee2800000 pthread_self is 476229647696 exec_env->native_stack_boundary is 0x7ff708d200 &prev_frame is 0x6ee17f9370
    2021-04-20 10:57:15.485 28264-28431 I/wamr-manager: xingkaiyu-test android native call wasm_runtime_module_dup_data exception module_inst is 0x6f4894c480 exception is Exception: native stack overflow_

xujuntwt95329, 2021-04-20:
Thanks for the information, we've moved `wasm_exec_env_set_thread_info` into lower layer in PR #617 , the thread info will be set before execution, so that the `exec_env` is always bound to the current execution thread. Could you please pull the code and try again?

closed 2021-04-21 by xingkaiyu