wasm-micro-runtime/wasm-micro-runtime #603: Question about pthread lib, the thread cannot return right value when joined

bxkai, 2021-04-08:
hi， I create two thread，the thread exited and return value to main join thread.  The first thread return rightly, but the second thread is error.

main.c：
#include <stdio.h>
#include <stdlib.h>
#include "pthread.h"

static void *ThreadRoutine(void *arg)
{
    int *a = (int*)malloc(sizeof(int));
    *a = 66666;
    printf( "%s\n", (char*)arg);
    pthread_exit((void*)a);
    return NULL;
}

int TwoPthreadCreate(void)
{
    pthread_t id1, id2;
    void *a1, *a2;
    int i, ret1, ret2;
    char s1[] = "This is first thread!";
    char s2[] = "This is second thread!";

    ret1 = pthread_create(&id1, NULL, ThreadRoutine, s1);
    if (ret1 != 0) {
        printf("Create pthread1 error! %d\n", ret1);
    }

    ret2 = pthread_create(&id2, NULL, ThreadRoutine, s2);
    if (ret2 != 0) {
        printf( "Create pthread2 error! %d\n", ret2);
    }

    ret1 = pthread_join(id1, &a1);
    if (ret1 != 0) {
        printf("Join pthread1 error! %d \n", ret1);
    }

    ret2 = pthread_join(id2, &a2);
    if (ret2 != 0) {
        printf( "Join pthread2 error! %d\n", ret2);
    }

    printf( "thread1 return %d\n", *(int*)a1);
    printf( "thread2 return %d\n", *(int*)a2);
    return 0;
}

1、build and run with native gcc：
        gcc main.c -lpthread
        ./a.out 
        This is second thread!
        This is first thread!
        thread1 return 66666
        thread2 return 66666

2、build and run with wasm
     1） build wasm
            /opt/wasi-sdk/bin/clang -pthread -O3                \
                  -Wl,--shared-memory,--max-memory=196608         \
                  -Wl,--allow-undefined,--no-check-features       \
                  -Wl,--export=__heap_base,--export=__data_end    \
                  main.c -o test.wasm
     2） run with iwasm：
            ./iwasm test.wasm 
            THREAD CREATED 0x7fbb1b323ed0
            This is first thread!
            THREAD CREATED 0x7fbb1ae50ed0
            [00:44:54:513 - 1AE97740]: HashMap find elem failed: map or key is NULL.
            
            thread1 return 66666
            thread2 return 0

      The thread2 return 0, not as expected. Please to reslove the question, Thanks!

xujuntwt95329, 2021-04-08:
Thanks for reporting this issue, it is caused by this line:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/77c3ddf7d0d946258cbf56ea6c9b1c14e46e4061/core/iwasm/libraries/lib-pthread/lib_pthread_wrapper.c#L651
the pointer size is 32bit in wasm but 64bit in host, so in the first `pthread_join`, four more bytes are written which by accident modify the `id2` variable in your source code.

You can change this line to
``` C
*(uint32*)retval = (uint32)(uintptr_t)ret;
```
Then you'll get correct result. We'll submit a pull request to fix this later.

BTW, the `ThreadRoutine` function in your source code seems not correct, the compiler throw several errors when I try to compile, so I modify some lines, the modified code is:
``` C
static void *ThreadRoutine(void *arg)
{
    int *a = (int *)malloc(sizeof(int));
    *a = 66666;
    printf("%s\n", (char *)arg);
    pthread_exit((void *)a);
    return NULL;
}
```

bxkai, 2021-04-09:
hi, As your change, The result is right.  The ThreadRoutine  function is change wrongly when I coping code. Thanks!

closed 2021-04-09 by bxkai