wasm-micro-runtime/wasm-micro-runtime #607: Question about AssemblyScript version support

AfflatusX, 2021-04-09:
I upgraded my AS WASM app to 0.18.23 (latest AssemblyScript version), it compiles/installs fine, but timer no longer works (it works fine on 0.17.15).

1. Any idea on what could possibly cause that?
2. What should we expect in terms of version compatibilty between WAMR and AssemblyScript in general? Is it possible to create some CI to make sure things don't break?

xujuntwt95329, 2021-04-20:
Hi, I tested the `timer` sample on latest `AssemblyScript`, and it works well
``` bash
2 seconds passed
4 seconds passed
6 seconds passed
8 seconds passed
10 seconds passed
12 seconds passed
14 seconds passed
16 seconds passed
18 seconds passed
20 seconds passed
Stop Timer
```

The `asc` version is : `Version 0.18.27`

Have you modified the code in your local repo ?

AfflatusX, 2021-05-20:
actually I just tried to upgrade WAMR to latest (4-15-2021) and tried again in AoT mode (I didn't modify WAMR code).

It crashed the system (I'm running in QEMU with MPS2-AN385 board). The same WASM code works fine with 0.17.5 ASC version.

AfflatusX, 2021-05-20:
turned out I have to re-compile the wamrc. Both the timer and AS code is running properly now.

A follow up question - I noticed 0.18 asc binary consumes about 15% more memory than 0.17. Is that something you guys are aware of?

lum1n0us, 2026-04-14:
Thanks for raising this! We don't have detailed compatibility data for AssemblyScript 0.18.x or later at the moment. Feel free to give it a try and see how it goes — we'd love to hear what you discover, whether it works smoothly or you run into any quirks!